Rdiger-36/HaspelSync #4: Feature Request

zacgoodwin, 2024-12-17:
It would be super neat if there was an option to have filamentstatus create new spools for you in spoolman.

The tricky part is that the spools have two tags. A couple approaches could be taken here when there is no direct tag match:

- Since Color Code, Material Type and % of usage is present if the check doesn't find a spool with that tag, then it looks for a spool with similar properties (color, material, %) and is missing a secondary tag then updates that spool to have the secondary tag of the new spool in question. (see attached image).

[image]

- If there is no tag match have a "queue" in the spoolman interface (if possible) that you can go and either merge it with an existing spool or have it create a new spool. Could be a simple webpage in the filamentstatus container too

Rdiger-36, 2024-12-17:
This looks good and its a good idea. But it could take some time.
Im not really satisfied with the JavaScript solution so i could rewrite it and also implement a little webpage.

The problem with the secondary tag is solved.
There is another unique tag tray_uuid.
The tray_uuid is in all 4 slots the only unique number for 1 spool. It doesn't matter in which slot the spool is located or if there where other spools in this slot before.
I have tested it with all my 23 spools available. I also turned my printer off and on and also pulled the plug. The tray_uuid still stays the same. I always thougt this is the number of the RFID reader. After that i changed the Script back to one tag

zacgoodwin, 2024-12-17:
Do you know if the tray uuid stays the same between AMSes? 

I have a lite here along side my X1C I can test with if needed :)

Rdiger-36, 2024-12-18:
Mh no, unfortunately not.
I only have one AMS with my P1S, so i don't have the chance to test it.
It would be really nice if you could test it.

Yeah thats a good idea to use multiple printers with AMS in one container, i think about it, how i could implement it.

zacgoodwin, 2024-12-18:
Looks like the AMS Lite only will tell you what is loaded not the % remaining but the ID does stay the same. 

[image]

Rdiger-36, 2024-12-18:
Ok, thank you for your research

zacgoodwin, 2024-12-18:
I have 2 full AMS and the ID does transfer between them

port513, 2024-12-26:
This is a feature I would love to have.
I have 2 AMS at the moment.

Not sure how much time I can spend with developing but I can for sure test things out if needed.

Rdiger-36, 2024-12-26:
Hi @port513 at the moment I am working on the web UI. 
There you can migrate or create spools with one button click. In addition, a filament should be created if it is not yet stored in Spoolman.
I am finished with the migration process. With the creation of the spools incl. filament creation if necessary, I am at about 50%.

If all this works, then the implementation for automatic creation and migration of the spools are no longer a big problem and should be relatively easy to set up

port513, 2024-12-26:
> Hi @port513 at the moment I am working on the web UI. There you can migrate or create spools with one button click. In addition, a filament should be created if it is not yet stored in Spoolman. I [...]

Sound nice, I will setup spoolman with postgres and your integration and wait for testing then 😁

Rdiger-36, 2024-12-29:
I have implemented some ideas and tested it a little bit. It's not ready at all but I think I can show you a preview. With the tag :dev you can test the new container with a web implementation and an automated spool creation and merging. For that you can add MODE as an ENV with 2 options: "manual" or "automatic" (standard is manual).
The port for this container is 4000

Example docker-compose.yml:

```bash
services:
  bambuspoolman-dev:
    image: ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:dev
    container_name: bambuspoolman-dev
    ports:
      - 4000:4000
    environment:
      - PRINTER_ID=CHANGE_ME
      - PRINTER_CODE=CHANGE_ME
      - PRINTER_IP=CHANGE_ME
      - SPOOLMAN_IP=CHANGE_ME
      - SPOOLMAN_PORT=CHANGE_ME
      - UPDATE_INTERVAL=60000
      - MODE=automatic
    restart: unless-stopped
```
Important is, that your filament type in Spoolman is set by the filament name requested from the AMS:

## Spoolman
[image]

## AMS Data
[image]

L0rdS474n, 2024-12-30:
Thank you! I will proceed with running it from now on. If you require assistance with any additional testing, please feel free to let me know.

Rdiger-36, 2024-12-30:
Hi @L0rdS474n how does your envs look like?
Is the website accessable?
If yes then please look in the dev tools from your web browser for any errors

L0rdS474n, 2024-12-30:
> Hi @L0rdS474n how does your envs look like? Is the website accessable? If yes then please look in the dev tools from your web browser for any errors.
> 
> When i doesn't deploy the website, it also doesn't work. I have a new idea for the logic, but i have to do some work on it. I will tell you when i have an update for the dev container

Yes, the website is up and say it's connected to the printer. I'll check for error later today.

Rdiger-36, 2024-12-30:
> Yes, the website is up and say it's connected to the printer. I'll check for error later today.

Sry was my mistake. I implemented the logic into the frontend. I will pack it to the backend

L0rdS474n, 2024-12-30:
> > Yes, the website is up and say it's connected to the printer. I'll check for error later today.
> 
> Sry was my mistake. I implemented the logic into the frontend. I will pack it to the backend

The setup is now fully functional, and it automatically adds spools as needed!
For now, I’ll keep running the development version. If there’s anything you’d like to test, let me know, and I’ll restart the containers to use the latest updates.

If anyone is interested in deploying this setup with PostgreSQL, here’s how you can do it:

Create a new folder.
Add a docker-compose.yaml file in the folder with the following content:

```
services:
  db:
    image: postgres
    restart: unless-stopped
    shm_size: 128mb
    volumes:
      - ./data/db:/var/lib/postgresql/data # Persistent storage for database
    environment:
      - POSTGRES_PASSWORD=<YOUR POSTGRES PASSWORD>
      - POSTGRES_USER=<YOUR POSTGRES USER>
      - POSTGRES_DB=<YOUR DATABASE NAME>

  adminer:
    image: adminer
    restart: unless-stopped
    ports:
      - "8080:8080"

  spoolman:
    image: ghcr.io/donkie/spoolman:latest
    restart: unless-stopped
    depends_on:
      - db
    volumes:
      # Mount the host machine's ./data directory into the container's /home/app/.local/share/spoolman directory
      - type: bind
        source: ./data # This is where the data will be stored locally. Could also be set to for example `source: /home/pi/printer_data/spoolman`.
        target: /home/app/.local/share/spoolman # Do NOT modify this line
    ports:
      # Map the host machine's port 7912 to the container's port 8000
      - "7912:8000"
    environment:
      - SPOOLMAN_DB_TYPE=postgres
      - SPOOLMAN_DB_HOST=db
      - SPOOLMAN_DB_PORT=5432
      - SPOOLMAN_DB_NAME=<YOUR DATABASE NAME>
      - SPOOLMAN_DB_USERNAME=<YOUR POSTGRES USER>
      - SPOOLMAN_DB_PASSWORD=<YOUR POSTGRES PASSWORD>
      - SPOOLMAN_LOGGING_LEVEL=INFO
      - TZ=Europe/Stockholm # Optional, defaults to UTC

  bambuspoolman-dev:
    image: ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:dev
    container_name: bambuspoolman-dev
    depends_on:
      - spoolman
    ports:
      - 4000:4000
    environment:
      - PRINTER_ID=<YOUR PRINTER SERIAL>
      - PRINTER_CODE=<YOUR PRINTER ACCESS CODE>
      - PRINTER_IP=<YOUR PRINTER IP>
      - SPOOLMAN_IP=<YOUR SPOOLMAN IP THAT IS ACCESIBLE ON YOUR NETWORK>
      - SPOOLMAN_PORT=7912
      - UPDATE_INTERVAL=60000
      - MODE=automatic
    restart: unless-stopped

```
This configuration sets up a PostgreSQL database with Adminer, Spoolman, and the development version of bambulab-ams-spoolman-filamentstatus.
The PostgreSQL database is stored locally under ./data/db, making it easy to back up alongside the docker-compose.yaml file.

darkorb, 2025-01-02:
I've also setup this with my Spoolman and it's working well so far - thanks for your work on this to date.

Happy to do more testing of things if needed.

Rdiger-36, 2025-01-02:
Thank you all for you support ❤️

The next dev test Build is allmost done and i will push it as soon as possible.
I will tell you when you can test it.

Rdiger-36, 2025-01-02:
I published the new dev build that contains much more improvements and some fixes.
Please let me know if something is wiring or missed

Rdiger-36, 2025-01-04:
I published the new Update. The dev image will be deleted, only use latest

closed 2025-01-04 by Rdiger-36