Rdiger-36/HaspelSync #7: No AMS updates showing in logs

sundyap, 2025-01-02:
I don't get any AMS updates in the logs. All i get is the folowing:
`MQTT client connected
Subscribed to device/01S00C431800472/report
Waiting for MQTT messages...`

Even tried de dev build, and  nothing.

Rdiger-36, 2025-01-02:
Hi, how does your ENVs Looks Like?
What is your update interval?

sundyap, 2025-01-02:
This is my ENV's:
`environment:
      - PRINTER_ID=01S00C431800472
      - PRINTER_CODE=myaccescode
      - PRINTER_IP=printerip
      - SPOOLMAN_IP=spoolmanip
      - SPOOLMAN_PORT=7912
      - UPDATE_INTERVAL=1200`

Rdiger-36, 2025-01-02:
Mh ok, can you test it with the new dev version?

sundyap, 2025-01-02:
Here are the logs for de dev build:
`Backend running on http://localhost:4000
MQTT client connected
Subscribed to device/01S00C431800472/report
healthy
Waiting for MQTT messages...
AMS [A] (hum: 5, temp: 0.0ºC)
AMS [A] (hum: 5, temp: 0.0ºC)
AMS [A] (hum: 5, temp: 0.0ºC)
AMS [A] (hum: 5, temp: 0.0ºC)
`

Rdiger-36, 2025-01-02:
Ok, on which system do you run the Docker container?

Can you manualy request the MQTT Connection data and Share it?

On Linux you can use the following command:

First create a certificate and then Connect to the Printer:

```bash

openssl genrsa -out ca.key 2048

openssl req -new -x509 -days 3650 -key ca.key -out blcert.pem

mosquitto_sub -h PRINTER_IP -p 8883 --cafile ./blcert.pem --insecure -u bblp -P PRINTER_ACCESCODE -t device/PRINTER_SERIAL/report

```

luis15pt, 2025-01-02:
Using the new dev version and its working perfectly. 
```
bambulab-ams-spoolman-filamentstatus  | Checking Extra Field "tag"...
bambulab-ams-spoolman-filamentstatus  | Spoolman Extra Field "tag" for Spool is set: true
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  | Backend running on http://localhost:4000
bambulab-ams-spoolman-filamentstatus  | MQTT client connected
bambulab-ams-spoolman-filamentstatus  | Subscribed to device/01P09A451702885/report
bambulab-ams-spoolman-filamentstatus  | Waiting for MQTT messages...
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  | AMS [A] (hum: 2, temp: 0.0ºC)
bambulab-ams-spoolman-filamentstatus  |     - [A0] PLA Matte A3D8E1FF (89%) [[ A10249B446794734A74E9ADC963161EC ]]
bambulab-ams-spoolman-filamentstatus  |         - A new Filament and a new Spool can be created with following attributes:
bambulab-ams-spoolman-filamentstatus  |           New Filament => Manufacturer: Bambu Lab, Material: PLA, Color: Matte Ice Blue
bambulab-ams-spoolman-filamentstatus  |           creating Filament and Spool...
bambulab-ams-spoolman-filamentstatus  |           Filament and Spool successfully created for Spool in AMS Slot => A0!
bambulab-ams-spoolman-filamentstatus  |     - [A1] PLA Basic FFFFFFFF (33%) [[ B6C2F91AC7F440F095D0C0E0C05203F8 ]]
bambulab-ams-spoolman-filamentstatus  |         - A new Filament and a new Spool can be created with following attributes:
bambulab-ams-spoolman-filamentstatus  |           New Filament => Manufacturer: Bambu Lab, Material: PLA, Color: Jade White
bambulab-ams-spoolman-filamentstatus  |           creating Filament and Spool...
bambulab-ams-spoolman-filamentstatus  |           Filament and Spool successfully created for Spool in AMS Slot => A1!
bambulab-ams-spoolman-filamentstatus  |     - [A2] PLA Basic 00AE42FF (100%) [[ 51BE582A85764C47B55012A7C2BDB4C8 ]]
bambulab-ams-spoolman-filamentstatus  |         - A new Filament and a new Spool can be created with following attributes:
bambulab-ams-spoolman-filamentstatus  |           New Filament => Manufacturer: Bambu Lab, Material: PLA, Color: Bambu Green
bambulab-ams-spoolman-filamentstatus  |           creating Filament and Spool...
bambulab-ams-spoolman-filamentstatus  |           Filament and Spool successfully created for Spool in AMS Slot => A2!
bambulab-ams-spoolman-filamentstatus  |     - [A3] PLA Matte 68724DFF (46%) [[ 1CB08587CB33496FA7C48CA5E63E81B6 ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 4 => Matte Dark Green
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  | AMS [B] (hum: 2, temp: 0.0ºC)
bambulab-ams-spoolman-filamentstatus  |     - [B0] PLA Basic F7E6DEFF (70%) [[ BB4243E2578E4A0D86657FDEFB9A6071 ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 5 => Beige
bambulab-ams-spoolman-filamentstatus  |     - [B1] PLA Basic FF6A13FF (1%) [[ 52770395E6E44FB585320A51CB4CEBB8 ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 6 => Orange
bambulab-ams-spoolman-filamentstatus  |     - [B2] PLA Basic 000000FF (2%) [[ E0943EFEED914CE1BA9A41F015EEF67E ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 7 => Black
bambulab-ams-spoolman-filamentstatus  |     - [B3] PLA Matte 0078BFFF (85%) [[ ED2F80B7ADE940C0A082477BCA0010AB ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 8 => Matte Marine Blue
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  | AMS [A] (hum: 2, temp: 0.0ºC)
bambulab-ams-spoolman-filamentstatus  |     - [A0] PLA Matte A3D8E1FF (89%) [[ A10249B446794734A74E9ADC963161EC ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 9 => Matte Ice Blue
bambulab-ams-spoolman-filamentstatus  |     - [A1] PLA Basic FFFFFFFF (34%) [[ B6C2F91AC7F440F095D0C0E0C05203F8 ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 10 => Jade White
bambulab-ams-spoolman-filamentstatus  |     - [A2] PLA Basic 00AE42FF (100%) [[ 51BE582A85764C47B55012A7C2BDB4C8 ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 11 => Bambu Green
bambulab-ams-spoolman-filamentstatus  |     - [A3] PLA Matte 68724DFF (46%) [[ 1CB08587CB33496FA7C48CA5E63E81B6 ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 4 => Matte Dark Green
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  | AMS [B] (hum: 2, temp: 0.0ºC)
bambulab-ams-spoolman-filamentstatus  |     - [B0] PLA Basic F7E6DEFF (70%) [[ BB4243E2578E4A0D86657FDEFB9A6071 ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 5 => Beige
bambulab-ams-spoolman-filamentstatus  |     - [B1] PLA Basic FF6A13FF (1%) [[ 52770395E6E44FB585320A51CB4CEBB8 ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 6 => Orange
bambulab-ams-spoolman-filamentstatus  |     - [B2] PLA Basic 000000FF (2%) [[ E0943EFEED914CE1BA9A41F015EEF67E ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 7 => Black
bambulab-ams-spoolman-filamentstatus  |     - [B3] PLA Matte 0078BFFF (85%) [[ ED2F80B7ADE940C0A082477BCA0010AB ]]
bambulab-ams-spoolman-filamentstatus  |         - Updated Spool-ID 8 => Matte Marine Blue
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  | AMS [A] (hum: 2, temp: 0.0ºC)
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  |
bambulab-ams-spoolman-filamentstatus  | AMS [B] (hum: 2, temp: 0.0ºC)
bambulab-ams-spoolman-filamentstatus  |

```

Rdiger-36, 2025-01-04:
I have implemented some fixes and new logic to the latest image.
The dev will be deleted.

Please check if this help you

luis15pt, 2025-01-04:
Does it have all the stuff from dev ?

Rdiger-36, 2025-01-04:
@luis15pt 
Yes and some additional improvements

Rdiger-36, 2025-01-08:
Mh ok, this only could be when the service couldn't get valid data fr the printer. For this I need some info what the printer sends.

Did you tried to restart your printer?

> Additionally, I tried running the commands that you suggested above and received the following after the 3rd line:
> 
> zsh: command not found: mosquitto_sub
> 

This only works on Linux. For Mac with Apple Silicon you have to install homebrew and over that you must install mosquitto with this command: 
```bash
brew install mosquitto
```
And then the command should work.

Rdiger-36, 2025-01-08:
There is some output, wait for a output that should look like this (maybe you have to trigger this information by open Bambu Studio or Bambu Handy)

```bash

{
  "print": {
    "upgrade_state": {
      "sequence_id": 0,
      "progress": "",
      "status": "IDLE",
      "consistency_request": false,
      "dis_state": 0,
      "err_code": 0,
      "force_upgrade": false,
      "message": "0%, 0B/s",
      "module": "",
      "new_version_state": 2,
      "cur_state_code": 0,
      "idx2": 2308936552,
      "new_ver_list": []
    },
    "ipcam": {
      "ipcam_dev": "1",
      "ipcam_record": "disable",
      "timelapse": "disable",
      "resolution": "",
      "tutk_server": "enable",
      "mode_bits": 3
    },
    "upload": {
      "status": "idle",
      "progress": 0,
      "message": ""
    },
    "net": {
      "conf": 0,
      "info": [
        {
          "ip": 436316352,
          "mask": 16777215
        }
      ]
    },
    "nozzle_temper": 34.625,
    "nozzle_target_temper": 0,
    "bed_temper": 35.34375,
    "bed_target_temper": 0,
    "chamber_temper": 5,
    "mc_print_stage": "1",
    "heatbreak_fan_speed": "0",
    "cooling_fan_speed": "0",
    "big_fan1_speed": "0",
    "big_fan2_speed": "0",
    "mc_percent": 100,
    "mc_remaining_time": 0,
    "ams_status": 0,
    "ams_rfid_status": 0,
    "hw_switch_state": 0,
    "spd_mag": 100,
    "spd_lvl": 2,
    "print_error": 0,
    "lifecycle": "product",
    "wifi_signal": "-70dBm",
    "gcode_state": "FINISH",
    "gcode_file_prepare_percent": "100",
    "queue_number": 0,
    "queue_total": 0,
    "queue_est": 0,
    "queue_sts": 0,
    "project_id": "210131144",
[... 170 lines cut ...]
      "tray_temp": "0",
      "tray_time": "0",
      "bed_temp_type": "0",
      "bed_temp": "0",
      "nozzle_temp_max": "0",
      "nozzle_temp_min": "0",
      "xcam_info": "000000000000000000000000",
      "tray_uuid": "00000000000000000000000000000000",
      "remain": 0,
      "k": 0.019999999552965164,
      "n": 1,
      "cali_idx": -1
    },
    "lights_report": [
      {
        "node": "chamber_light",
        "mode": "off"
      }
    ],
    "command": "push_status",
    "msg": 0,
    "sequence_id": "5547"
  }
}
```

Rdiger-36, 2025-01-09:
I released a new Update, maybe it could also fixed your problems

buffdriver, 2025-01-09:
Thank you for your continued work on this!  I pulled the new package and the results are the same. FWIW, I should tell you that the docker launch command in the README was not publishing port 4000, so I have been launching with a slightly modified command:

docker run -d -p 127.0.0.1:4000:4000 \
  -e PRINTER_ID=01P09C4925XXXXX \
  -e PRINTER_CODE=3000XXXX \
  -e PRINTER_IP=192.168.5.155 \
  -e SPOOLMAN_IP=192.168.5.172 \
  -e SPOOLMAN_PORT=8000 \
  -e UPDATE_INTERVAL=120000 \
  --name bambulab-ams-spoolman-filamentstatus \
 ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:latest

Do you see any issues with this?  Also, I installed mosquitto with homebrew and received the following:
mark@Dumbledore ~ % mosquitto_sub -h 192.168.5.155 -p 8883 --cafile ./blcert.pem --insecure -u bblp -P 3000XXXX -t device/01P09C4925XXXXX/report
Error: Protocol error

buffdriver, 2025-01-09:
Here is a screenshot of the frontend

[image]

Rdiger-36, 2025-01-09:
Yeah its also in my interest that every one could use this service.

I have implemented a output for all incoming mqtt messages in the dev build:
ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:dev

Can you please use this and look at the logs?
it should looks like this:
```bash
[LOG] 2025-01-09T16:37:36.416Z - Setting up configuration...
[LOG] 2025-01-09T16:37:36.428Z - Spoolman connection: true
[LOG] 2025-01-09T16:37:36.428Z - Checking Vendors...
[LOG] 2025-01-09T16:37:36.434Z - Vendor "Bambu Lab" exists: true
[LOG] 2025-01-09T16:37:36.434Z - Checking Extra Field "tag"...
[LOG] 2025-01-09T16:37:36.436Z - Spoolman Extra Field "tag" for Spool is set: true
[LOG] 2025-01-09T16:37:36.436Z - Backend running on http://localhost:4000
[LOG] 2025-01-09T16:37:36.437Z - Setting up MQTT connection...
[LOG] 2025-01-09T16:37:37.313Z - MQTT client connected
[LOG] 2025-01-09T16:37:37.328Z - Subscribed to device/01P00XXXXXXX/report
[LOG] 2025-01-09T16:37:37.328Z - Waiting for MQTT messages...
[LOG] 2025-01-09T16:37:39.303Z - {"print":{"wifi_signal":"-69dBm","command":"push_status","msg":1,"sequence_id":"25010"}}
[LOG] 2025-01-09T16:37:41.356Z - {"print":{"bed_temper":13.46875,"command":"push_status","msg":1,"sequence_id":"25011"}}
[LOG] 2025-01-09T16:37:47.489Z - {"print":{"bed_temper":13.5,"command":"push_status","msg":1,"sequence_id":"25012"}}
[LOG] 2025-01-09T16:37:49.508Z - {"print":{"bed_temper":13.46875,"command":"push_status","msg":1,"sequence_id":"25013"}}
[LOG] 2025-01-09T16:37:51.362Z - {"print":{"upgrade_state":{"sequence_id":0,"progress":"","status":"IDLE","consistency_request":false,"dis_state":0,"err_code":0,"force_upgrade":false,"message":"0%, 0B/s","module":"","new_version_state":2,"cur_state_code":0,"idx2":2308936552,"new_ver_list":[]},"ipcam":{"ipcam_dev":"1","ipcam_record":"disable","timelapse":"disable","resolution":"","tutk_server":"enable","mode_bits":3},"upload":{"status":"idle","progress":0,"message":""},"net":{"conf":0,"info":[{"ip":436316352,"mask":16777215}]},"nozzle_temper":16.28125,"nozzle_target_temper":0,"bed_temper":13.4375,"bed_target_temper":0,"chamber_temper":5,"mc_print_stage":"1","heatbreak_fan_speed":"0","cooling_fan_speed":"0","big_fan1_speed":"0","big_fan2_speed":"0","mc_percent":100,"mc_remaining_time":0,"ams_status":0,"ams_rfid_status":0,"hw_switch_state":0,"spd_mag":100,"spd_lvl":2,"print_error":0,"lifecycle":"product","wifi_signal":"-69dBm","gcode_state":"FINISH","gcode_file_prepare_percent":"100","queue_number":0,"queue_total":0,"queue_est":0,"queue_sts":0,"project_id":"210131144","profile_id":"197179333","task_id":"423071206","subtask_id":"423071207","subtask_name":"(Nicht_gespeichert)","gcode_file":"","stg":[],"stg_cur":255,"print_type":"idle","home_flag":24331672,"mc_print_line_number":"0","mc_print_sub_stage":0,"sdcard":true,"force_upgrade":false,"mess_production_state":"active","layer_num":7,"total_layer_num":7,"s_obj":[],"filam_bak":[],"fan_gear":0,"nozzle_diameter":"0.4","nozzle_type":"hardened_steel","cali_version":0,"k":"0.0200","flag3":15,"hms":[],"online":{"ahb":false,"rfid":false,"version":2135901539},"ams":{"ams":[{"id":"0","humidity":"5","temp":"0.0","tray":[{"id":"0","remain":16,"k":0.019999999552965164,"n":1,"cali_idx":-1,"tag_uid":"AA7C6CDC00000100","tray_id_name":"G02-K0","tray_info_idx":"GFG02","tray_type":"PETG","tray_sub_brands":"PETG HF","tray_color":"000000FF","tray_weight":"1000","tray_diameter":"1.75","tray_temp":"65","tray_time":"8","bed_temp_type":"0","bed_temp":"0","nozzle_temp_max":"260","nozzle_temp_min":"230","xcam_info":"8813A438F40158020000803F","tray_uuid":"1CEC14C7DB18404FB71B61DBC4549322","ctype":0,"cols":["000000FF"]},{"id":"1","remain":11,"k":0.019999999552965164,"n":1,"cali_idx":-1,"tag_uid":"3EC8F47300000100","tray_id_name":"A00-K0","tray_info_idx":"GFA00","tray_type":"PLA","tray_sub_brands":"PLA Basic","tray_color":"000000FF","tray_weight":"1000","tray_diameter":"1.75","tray_temp":"55","tray_time":"8","bed_temp_type":"0","bed_temp":"0","nozzle_temp_max":"230","nozzle_temp_min":"190","xcam_info":"803E803EE803E803CDCC4C3F","tray_uuid":"E6790170348E4538973BA8D16D841273","ctype":0,"cols":["000000FF"]},{"id":"2","remain":15,"k":0.019999999552965164,"n":1,"cali_idx":-1,"tag_uid":"5AB2F0AF00000100","tray_id_name":"A00-W1","tray_info_idx":"GFA00","tray_type":"PLA","tray_sub_brands":"PLA Basic","tray_color":"FFFFFFFF","tray_weight":"1000","tray_diameter":"1.75","tray_temp":"55","tray_time":"8","bed_temp_type":"0","bed_temp":"0","nozzle_temp_max":"230","nozzle_temp_min":"190","xcam_info":"34218813F401E8030000003F","tray_uuid":"E1B79435D82A44A09C44828DF941EB90","ctype":0,"cols":["FFFFFFFF"]},{"id":"3","remain":100,"k":0.019999999552965164,"n":1,"cali_idx":-1,"tag_uid":"CAC4FDE900000100","tray_id_name":"A00-A0","tray_info_idx":"GFA00","tray_type":"PLA","tray_sub_brands":"PLA Basic","tray_color":"FF6A13FF","tray_weight":"1000","tray_diameter":"1.75","tray_temp":"55","tray_time":"8","bed_temp_type":"0","bed_temp":"0","nozzle_temp_max":"230","nozzle_temp_min":"190","xcam_info":"8813100EE803E8039A99193F","tray_uuid":"93DCD12A3F9149BCBF57317D73BB737C","ctype":0,"cols":["FF6A13FF"]}]}],"ams_exist_bits":"1","tray_exist_bits":"f","tray_is_bbl_bits":"f","tray_tar":"255","tray_now":"255","tray_pre":"255","tray_read_done_bits":"f","tray_reading_bits":"0","version":42,"insert_flag":true,"power_on_flag":true},"vt_tray":{"id":"254","tag_uid":"0000000000000000","tray_id_name":"","tray_info_idx":"","tray_type":"","tray_sub_brands":"","tray_color":"0 [... cut ...] 0000000000000000000000","tray_uuid":"00000000000000000000000000000000","remain":0,"k":0.019999999552965164,"n":1,"cali_idx":-1},"lights_report":[{"node":"chamber_light","mode":"off"}],"command":"push_status","msg":0,"sequence_id":"25015"}}
```

 
What we need is the long message. This contains all the data we need to process.
Please look through your Logs, so you don't share something you didn't want.

I have added the port to the docker run command in my Readme, thank you for pointing this out.

buffdriver, 2025-01-09:
Running the dev build, my web page is unchanged.  However, there is plenty of info in the logs:

[logfile.log](https://github.com/user-attachments/files/18364670/logfile.log)

Rdiger-36, 2025-01-09:
Ok, the logs looks very good, its all collected right.

I will test it on my Mac mini M4 with docker and your docker run command

Rdiger-36, 2025-01-09:
Can you please try it with the new updated dev build?
On my Mac With M4 CPU it runs perfectly

sundyap, 2025-01-09:
i'm trying too, and it still doesn't work, but i get a new line in the log file:
`[LOG] 2025-01-09T17:29:42.643Z - No processable Data found for JSON filter data.printer.ams.ams`

buffdriver, 2025-01-09:
The new dev build logfile looks like the production:

2025-01-09 11:31:00 [LOG] 2025-01-09T17:31:00.158Z - Waiting for MQTT messages...
2025-01-09 11:33:03 [LOG] 2025-01-09T17:33:03.804Z - 
2025-01-09 11:33:03 [LOG] 2025-01-09T17:33:03.804Z - AMS [A] (hum: 5, temp: 0.0ºC)
2025-01-09 11:33:03 [LOG] 2025-01-09T17:33:03.805Z - 
2025-01-09 11:33:19 [LOG] 2025-01-09T17:33:19.796Z - No new AMS Data or changes in Spoolman found, wait for next Updates in 120 seconds ==> (09.01.2025 17:35:19)...

buffdriver, 2025-01-09:
Interesting followup:  I pulled the devbuild to a Debian Linux VM running in Parallels and launched it there.  I can access the front end, but there is no spool data there either.

Rdiger-36, 2025-01-09:
I have updated the dev build and added many debug outputs.
Can you please test it again?
Sometimes it helps to delete the docker image and pull it new.
Also please set your UPDATE_INTERVAL a bit lower, so its easier to debug this.
I use 15000 (15 sec.) for debugging, it's a good Intervall and it's does not stress Spoolman

buffdriver, 2025-01-09:
Log attached
[log.txt](https://github.com/user-attachments/files/18366228/log.txt)

sundyap, 2025-01-09:
Same for me as the log @buffdriver attached

Rdiger-36, 2025-01-09:
That looks good, this logs shows a lot.
The problem is the connection to the Spoolman integration.

Error in message handler: HTTPError: Response code 422 (Unprocessable Entity)

That means that Spoolman couldn't process the data send from my service.
That's the problem.

Whats your output from these APIs:

http://SPOOLMAN_IP:SPOOLMAN_PORT/api/v1/spools

http://SPOOLMAN_IP:SPOOLMAN_PORT/api/v1/field/spool

You can type these urls in your internet browser

sundyap, 2025-01-09:
http://SPOOLMAN_IP:SPOOLMAN_PORT/api/v1/spools
output:
`{
  "detail": "Not Found"
}`
http://SPOOLMAN_IP:SPOOLMAN_PORT/api/v1/field/spool
output:
`[
  {
    "name": "tag",
    "order": 1,
    "field_type": "text",
    "default_value": "\"\"",
    "key": "tag",
    "entity_type": "spool"
  }
]`

Rdiger-36, 2025-01-09:
I have a typo, the "s" is not there for the API call 

http://SPOOLMAN_IP:SPOOLMAN_PORT/api/v1/spool

sundyap, 2025-01-09:
ok, then this is the output:
`[]`

buffdriver, 2025-01-09:
http://SPOOLMAN_IP:SPOOLMAN_PORT/api/v1/spools
output:
[{"id":1,"registered":"2024-12-28T00:04:58Z","filament":{"id":1,"registered":"2024-12-27T23:54:41Z","name":"Green PLA","vendor":{"id":1,"registered":"2024-12-27T23:54:26Z","name":"Bambu Lab","empty_spool_weight":250.0,"external_id":"Bambu Lab","extra":{}},"material":"PLA","density":1.24,"diameter":1.75,"weight":1000.0,"spool_weight":250.0,"settings_extruder_temp":220,"settings_bed_temp":60,"color_hex":"00AE42","extra":{}},"price":18.39,"remaining_weight":1000.0,"initial_weight":1000.0,"spool_weight":250.0,"used_weight":0.0,"remaining_length":335283.6194167644,"used_length":0.0,"location":"Drybox","archived":false,"extra":{}},{"id":2,"registered":"2024-12-28T00:05:30Z","filament":{"id":2,"registered":"2024-12-27T23:55:20Z","name":"Black ABS","vendor":{"id":1,"registered":"2024-12-27T23:54:26Z","name":"Bambu Lab","empty_spool_weight":250.0,"external_id":"Bambu Lab","extra":{}},"material":"ABS","density":1.05,"diameter":1.75,"weight":1000.0,"spool_weight":250.0,"settings_extruder_temp":250,"settings_bed_temp":90,"color_hex":"000000","extra":{}},"price":18.39,"remaining_weight":1000.0,"initial_weight":1000.0,"spool_weight":250.0,"used_weight":0.0,"remaining_length":395953.9886445598,"used_length":0.0,"location":"Drybox","archived":false,"extra":{}},{"id":3,"registered":"2024-12-28T00:05:38Z","first_used":"2025-01-01T14:55:11Z","last_used":"2025-01-06T20:54:40Z","filament":{"id":3,"registered":"2024-12-27T23:56:20Z","name":"Black PETG-HF","vendor":{"id":1,"registered":"2024-12-27T23:54:26Z","name":"Bambu Lab","empty_spool_weight":250.0,"external_id":"Bambu Lab","extra":{}},"material":"PETG","density":1.25,"diameter":1.75,"weight":1000.0,"spool_weight":250.0,"settings_extruder_temp":250,"settings_bed_temp":70,"color_hex":"000000","extra":{}},"price":17.31,"remaining_weight":269.18999999999994,"initial_weight":1000.0,"spool_weight":250.0,"used_weight":730.8100000000001,"remaining_length":89532.95753071239,"used_length":243068.39293071788,"location":"Drybox","archived":false,"extra":{}},{"id":4,"registered":"2024-12-28T00:05:50Z","first_used":"2025-01-07T16:31:44Z","last_used":"2025-01-07T16:31:44Z","filament":{"id":4,"registered":"2024-12-27T23:56:45Z","name":"Black PLA","vendor":{"id":1,"registered":"2024-12-27T23:54:26Z","name":"Bambu Lab","empty_spool_weight":250.0,"external_id":"Bambu Lab","extra":{}},"material":"PLA","density":1.24,"diameter":1.75,"weight":1000.0,"spool_weight":250.0,"settings_extruder_temp":220,"settings_bed_temp":60,"color_hex":"000000","extra":{}},"price":17.31,"remaining_weight":435.0,"initial_weight":1000.0,"spool_weight":250.0,"used_weight":565.0,"remaining_length":145848.37444629252,"used_length":189435.24497047186,"location":"AMS Slot 1","archived":false,"extra":{"tag":"\"35A7F1AD1D634B6F93855182C3C17ABA\""}},{"id":5,"registered":"2024-12-28T00:06:01Z","filament":{"id":8,"registered":"2024-12-27T23:59:09Z","name":"Blue PLA","vendor":{"id":1,"registered":"2024-12-27T23:54:26Z","name":"Bambu Lab","empty_spool_weight":250.0,"external_id":"Bambu Lab","extra":{}},"material":"PLA","density":1.24,"diameter":1.75,"weight":1000.0,"spool_weight":250.0,"settings_extruder_temp":220,"settings_bed_temp":60,"color_hex":"0A2989","extra":{}},"price":17.31,"remaining_weight":1000.0,"initial_weight":1000.0,"spool_weight":250.0,"used_weight":0.0,"remaining_length":335283.6194167644,"used_length":0.0,"location":"Drybox","archived":false,"extra":{}},{"id":6,"registered":"2024-12-28T00:06:14Z","first_used":"2025-01-07T23:27:51Z","last_used":"2025-01-08T02:03:06Z","filament":{"id":11,"registered":"2024-12-28T00:01:28Z","name":"Gray PETG-HF","vendor":{"id":1,"registered":"2024-12-27T23:54:26Z","name":"Bambu Lab","empty_spool_weight":250.0,"external_id":"Bambu Lab","extra":{}},"material":"PETG","density":1.25,"diameter":1.75,"weight":1000.0,"spool_weight":250.0,"settings_extruder_temp":250,"settings_bed_temp":70,"color_hex":"9EA2A2","extra":{}},"price":18.39,"remaining_weight":349.0,"initial_weight":1000.0,"spool_weight":250.0,"used_weight":651.0,"remaining_length":116077.87131103917,"used_length":216523.4791503911,"location":"AMS Slot 3","archived":false,"extra":{"tag":"\"28FB98B3B5A64E0DAB5C3E18FCECEF7C\""}},{"id":7,"registered":"2024-12-28T00:06:23Z","filament":{"id":7,"registered":"2024-12-27T23:58:29Z","name":"Gray PLA","vendor":{"id":1,"registered":"2024-12-27T23:54:26Z","name":"Bambu Lab","empty_spool_weight":250.0,"external_id":"Bambu Lab","extra":{}},"material":"PLA","density":1.24,"diameter":1.75,"weight":1000.0,"spool_weight":250.0,"settings_extruder_temp":220,"settings_bed_temp":60,"color_hex":"8E9089","extra":{}},"price":17.31,"remaining_weight":1000.0,"initial_weight":1000.0,"spool_weight":250.0,"used_weight":0.0,"remaining_length":335283.6194167644,"used_length":0.0,"location":"Drybox","archived":false,"extra":{}},{"id":8,"registered":"2024-12-28T00:06:32Z","filament":{"id":6,"registered":"2024-12-27T23:57:50Z","name":"Orange PLA","vendor":{"id":1,"registered":"2024-12-27T23:54:26Z","name":"Bambu Lab","empty_spool_weight":250.0,"external_id":"Bambu Lab","extra":{}},"material":"PLA","density":1.24,"diameter":1.75,"weight":1000.0,"spool_weight":250.0,"settings_extruder_temp":220,"settings_bed_temp":60,"color_hex":"FF6A13","extra":{}},"price":14.06,"remaining_weight":1000.0,"initial_weight":1000.0,"spool_weight":250.0,"used_weight":0.0,"remaining_length":335283.6194167644,"used_length":0.0,"location":"Drybox","archived":false,"extra": [... cut ...] "diameter":1.75,"weight":1000.0,"spool_weight":152.0,"settings_extruder_temp":210,"settings_bed_temp":50,"color_hex":"898989","extra":{}},"price":0.0,"remaining_weight":58.0,"initial_weight":1000.0,"spool_weight":152.0,"used_weight":942.0,"remaining_length":18987.084967286373,"used_length":308376.44895144424,"location":"Drybox","archived":false,"extra":{}}]
http://SPOOLMAN_IP:SPOOLMAN_PORT/api/v1/field/spool
output:
[{"name":"tag","order":0,"field_type":"text","key":"tag","entity_type":"spool"}]

buffdriver, 2025-01-09:
I've got the issue narrowed down.  If the tag field is empty on all spools, it works. Once the spool and filament are created and saved to SpoolMan, future refreshes stop after encountering the first known filament.

Example:
Trays 0 - 3 are not in spoolman database, all four filament show in front end.

Now execute "create spool" for Tray 1 and refresh
Tray 0 shows
Tray 1 doesn't show because it's already there
Trays 2 & 3 don't show because processing stopped after encountering Tray 1

Rdiger-36, 2025-01-09:
@sundyap 
Can you test it with the new dev build and send me the logs?
I have added more debug logs

@buffdriver 
From your outputs it looks like that some spools doesn't got tag "field" or its empty like you said.
That could be your main Problem.
The Attribute for extra is empty, normally it must contain the Attribute "tag" and a empty value.
I didn't have the check if its a BambuLab spool, so I must add this check also to the code.
After this it should work properly

buffdriver, 2025-01-09:
I understand, and a quick question.  If there are additional extra fields beyond what your application requires, would that be a problem?  I originally had an extra field in spools called "active_tray".  Once I had problems, with your application, I deleted the extra fields and let them be recreated.

It would be nice if other additional fields were tolerated as long as the required ones were present.

Thanks!

Rdiger-36, 2025-01-09:
> I understand, and a quick question.  If there are additional extra fields beyond what your application requires, would that be a problem?  I originally had an extra field in spools called [...]

Yes you can use much more extra fields, the service only looks for the field called "tag" and adds this or update it.
If there are other fields it's no problem.

Rdiger-36, 2025-01-10:
Hi guys, I tried everything I can do and I think I cloud have found the issue.
Can you please test the new dev build?

buffdriver, 2025-01-10:
I'm getting a full display at port 4000.  Nice job, sir!  I also run an app called OpenSpoolMan which does RFID on non-bambu spools and the two seem to co-exist peacefully.  I would like it if Bambu would allow the % remaining to be delivered without changing the display in Bambu Studio, as I dislike it when only a sliver of color is displayed.  No worries though.  Thank you for your hard work and for sharing with the community!

Logfile attached.
[log.txt](https://github.com/user-attachments/files/18379882/log.txt)

Rdiger-36, 2025-01-10:
@buffdriver 
I am very happy that it works now.
Yes I also like the idea of OpenSpoolman and I am following the process of it.
Thank you for helping me to make this Servie a bit better

sundyap, 2025-01-10:
It's fine on my part. Thanks @Rdiger-36 for this greap app.

Rdiger-36, 2025-01-10:
@sundyap 
Thank you, then I can close this issue and I will publish the cleaned up version soon

Rdiger-36, 2025-01-10:
The update is now available in the latest build

closed 2025-01-10 by Rdiger-36

briodan, 2025-01-16:
just installed latest image in docker and I'm not getting any of the spool usage information. the Spools in the AMS have been detected and created in Spoolman but no usage is changed.

Attached logs below and here is my docker config

```
version: '3.8'
services:
 bambulab-ams-spoolman-filamentstatus:
   image: ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:latest
   container_name: bambulab-ams-spoolman-filamentstatus    
   ports:
      - 4000:4000
   environment:
     - PRINTER_ID=XXXXXX # Must be in capital letters!
     - PRINTER_CODE=XXXXX
     - PRINTER_IP=XX.XX.XX.XX
     - SPOOLMAN_IP=XX.XX.XX.YY
     - SPOOLMAN_PORT=7912
     - UPDATE_INTERVAL=120000 # Time in ms for updating spools in Spoolman (standard 120000 ms -> 2 minutes) min. 1000
     - DEBUG=true
     - MODE=automatic
     - SHOW_LOGS_WEB=true
   restart: unless-stopped
```

and screenshot from webui

[image]

[_bambulab-ams-spoolman-filamentstatus_logs (3).txt](https://github.com/user-attachments/files/18442684/_bambulab-ams-spoolman-filamentstatus_logs.3.txt)

buffdriver, 2025-01-16:
Have you turned on "Update remaining capacity" under AMS settings in Bambu Studio?

Rdiger-36, 2025-01-16:
Hi @buffdriver yes it is important for the App to turn on the "Update remaining capacity" option in Bambu Studio.
Sry that I didn't add this in my README. I will change it now.
I didn't know it isn't turned on by default, I must have turned it on when I got my printer and never looked at this setting again.

Can you please tell me if that option solved your problem?

briodan, 2025-01-16:
that took care of it, thanks all.

gislis2, 2025-01-20:
Apologies for commenting on a closed issue but maybe it helps other people. Spent 3 hours debugging until I realised:
Device ID is the SN, found under "Device info" on the X1C screen.
NOT the topmost "Device" field, in my case 3DP-XXX-XXX

After that everything works great. Thank you for this fantastic project.

Rdiger-36, 2025-01-20:
> Apologies for commenting on a closed issue but maybe it helps other people. Spent 3 hours debugging until I realised:
> Device ID is the SN, found under "Device info" on the X1C screen.
> NOT the [...]

Hi, thanks for your reply.
You are right, this could make some problems if you doesn't know it. I have updated the readme.