Rdiger-36/HaspelSync #8: Suggestion

luis15pt, 2025-01-02:
Could we add a feature to add the BambuLab Fillament reference number? for example Jade White is 10100, so when im placing an order i know exactly which refrences to order. 

[image]

luis15pt, 2025-01-02:
Nevermind saw this gets it from another repo and it doesnt contain it.
https://donkie.github.io/SpoolmanDB/filaments.json

closed 2025-01-02 by luis15pt

zacgoodwin, 2025-01-02:
> Nevermind saw this gets it from another repo and it doesnt contain it. https://donkie.github.io/SpoolmanDB/filaments.json

You can commit an update to that DB. I’ve been recording them under “Article Number” field personally.

Rdiger-36, 2025-01-06:
Hi @luis15pt if you commit an update to SpoolmanDB and donkie or his team include it, I can add this field to the program