Rdiger-36/HaspelSync #9: CF-Filaments are not being processed properly

ThatMattLatt, 2025-01-04:
Hello there,

first of all, thanks a lot for doing this connector. Just pulled up a container and set everything up, I noticed CF filaments (at least PLA-CF, PETG-CF, ASA-CF, maybe even more) do not get added to Spoolman when inserted into the AMS.

ASA gets fully recognized, PETG-CF does not:
[image]

Rdiger-36, 2025-01-04:
Hi, thank you. Can you please send me some data to see what data the AMS sends and how i can process it in the App?

This is how you can get the raw mqtt data from your printer, it you use Linux:

First of all you need to create a certificate and then Connect to your Printer via mosquitto_sub.

```bash
openssl genrsa -out ca.key 2048

openssl req -new -x509 -days 3650 -key ca.key -out blcert.pem

mosquitto_sub -h PRINTER_IP -p 8883 --cafile ./blcert.pem --insecure -u bblp -P PRINTER_ACCESCODE -t device/PRINTER_SERIAL/report
```

Rdiger-36, 2025-01-04:
I found the Problem:
Spoolman does not have these filaments in its Database.

You can See it here:

http://YOUR_SPOOLMAN_IP:YOUR_SPOOLMAN_PORT/api/v1/external/filament

ThatMattLatt, 2025-01-04:
Thanks for your prompt reply. Generated the certificate and tried to access it, got a "Protocol Error" in response:
[image]

ThatMattLatt, 2025-01-04:
Added the CF filaments in the SpoolmanDB repo and created a PR, will come back to this as soon as it's approved.

ThatMattLatt, 2025-01-06:
Works fine now. 

[image]

Closing the issue.

closed 2025-01-06 by ThatMattLatt