Rdiger-36/HaspelSync #11: Duplicate spools in logs, web UI & Spoolman creation

darkorb, 2025-01-09:
Hi,

Running the most recent release and have noticed that all the logs in the container are duplicating, as is the detection/display in the WebUI and also in the creation workflow - resulting in duplicates of new spools being created if they don't exist when put in the AMS (it's not duping pre-existing ones) in the automatic workflow.

Here's what happened in the log file - you can see it's detecting all the spools twice, not just new ones;

```
[LOG] 2025-01-09T04:48:54.182Z - AMS [A] (hum: 4, temp: 22.7ºC)
[LOG] 2025-01-09T04:48:54.207Z -     - [A0] PLA Basic 000000FF (62%) [[ 1B7073FB3041416BBB38B78743EAA291 ]]
[LOG] 2025-01-09T04:48:54.207Z -         - Updated Spool-ID 1 => Black
[LOG] 2025-01-09T04:48:54.306Z -     - [A0] PLA Basic 000000FF (62%) [[ 1B7073FB3041416BBB38B78743EAA291 ]]
[LOG] 2025-01-09T04:48:54.306Z -         - Updated Spool-ID 1 => Black
[LOG] 2025-01-09T04:48:54.339Z -     - [A1] PLA Basic 000000FF (100%) [[ ABAF3FE283F44940A4E36C22580768CE ]]
[LOG] 2025-01-09T04:48:54.339Z -         - Updated Spool-ID 3 => Black
[LOG] 2025-01-09T04:48:54.339Z -     - [A2] PETG HF 000000FF (100%) [[ 50F033E187D147DB9F0D53A22CC84C72 ]]
[LOG] 2025-01-09T04:48:54.340Z -         - A new Filament and Spool can be created:
[LOG] 2025-01-09T04:48:54.340Z -           Material: PETG, Color: HF Black
[LOG] 2025-01-09T04:48:54.340Z -           creating Filament and Spool...
[LOG] 2025-01-09T04:48:54.475Z -     - [A1] PLA Basic 000000FF (100%) [[ ABAF3FE283F44940A4E36C22580768CE ]]
[LOG] 2025-01-09T04:48:54.475Z -         - Updated Spool-ID 3 => Black
[LOG] 2025-01-09T04:48:54.476Z -     - [A2] PETG HF 000000FF (100%) [[ 50F033E187D147DB9F0D53A22CC84C72 ]]
[LOG] 2025-01-09T04:48:54.476Z -         - A new Filament and Spool can be created:
[LOG] 2025-01-09T04:48:54.476Z -           Material: PETG, Color: HF Black
[LOG] 2025-01-09T04:48:54.476Z -           creating Filament and Spool...
[LOG] 2025-01-09T04:48:54.576Z -           Filament and Spool successfully created for Spool in AMS Slot => A2!
[LOG] 2025-01-09T04:48:54.576Z - 
[LOG] 2025-01-09T04:48:54.702Z -           Filament and Spool successfully created for Spool in AMS Slot => A2!
[LOG] 2025-01-09T04:48:54.702Z - 
```

The WebUI has got a bit out of hand...
[image]

and here's the duplicated creation in Spoolman;

[image]

Any thoughts? Happy to test things as needed.

Rdiger-36, 2025-01-09:
Hi, it looks like that the Vendor Check isn't performed as expected.

Is there a Manufacturer in Spoolman called "Bambu Lab"?

If not, please create it by hand and restart you container and tell my of that fix your problem

darkorb, 2025-01-09:
Does seem to be there already;
[image]

Rdiger-36, 2025-01-09:
I have made some changes.
Can you please test it with the dev build?
ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:dev

darkorb, 2025-01-09:
Of course! Is running now.

Right away CPU usage looks much happier, I'll give it a little to settle into it's MQTT checking and then throw another spool in the AMS to see what happens.

Rdiger-36, 2025-01-09:
That sounds good.
I also looked through the code and saw a main problem that it doesn't block new Messages if the old ones not ready processed.

darkorb, 2025-01-09:
That appears to have resolved it, put a new spool in the AMS and it has been detected just once and created just one time in Spoolman. I'm now not seeing the duplicate messages on console either so all looks good to me.

```
[LOG] 2025-01-09T07:20:39.205Z - AMS [A] (hum: 4, temp: 26.2ºC)
[LOG] 2025-01-09T07:20:39.230Z -     - [A0] PLA Basic 000000FF (62%) [[ 1B7073FB3041416BBB38B78743EAA291 ]]
[LOG] 2025-01-09T07:20:39.230Z -         - Updated Spool-ID 1 => Black
[LOG] 2025-01-09T07:20:39.251Z -     - [A1] PLA Basic 000000FF (100%) [[ ABAF3FE283F44940A4E36C22580768CE ]]
[LOG] 2025-01-09T07:20:39.251Z -         - Updated Spool-ID 3 => Black
[LOG] 2025-01-09T07:20:39.253Z -     - [A2] PLA Basic BECF00FF (97%) [[ 815A328FCFCD41E0894AEF1BB18EAEB9 ]]
[LOG] 2025-01-09T07:20:39.253Z -         - A new Filament and Spool can be created:
[LOG] 2025-01-09T07:20:39.253Z -           Material: PLA, Color: Bright Green
[LOG] 2025-01-09T07:20:39.253Z -           creating Filament and Spool...
[LOG] 2025-01-09T07:20:39.300Z -           Filament and Spool successfully created for Spool in AMS Slot => A2!
[LOG] 2025-01-09T07:20:39.300Z - 
[LOG] 2025-01-09T07:21:39.711Z - 
[LOG] 2025-01-09T07:21:39.712Z - AMS [A] (hum: 4, temp: 26.2ºC)
[LOG] 2025-01-09T07:21:39.734Z -     - [A0] PLA Basic 000000FF (62%) [[ 1B7073FB3041416BBB38B78743EAA291 ]]
[LOG] 2025-01-09T07:21:39.734Z -         - Updated Spool-ID 1 => Black
[LOG] 2025-01-09T07:21:39.755Z -     - [A1] PLA Basic 000000FF (100%) [[ ABAF3FE283F44940A4E36C22580768CE ]]
[LOG] 2025-01-09T07:21:39.755Z -         - Updated Spool-ID 3 => Black
[LOG] 2025-01-09T07:21:39.775Z -     - [A2] PLA Basic BECF00FF (97%) [[ 815A328FCFCD41E0894AEF1BB18EAEB9 ]]
[LOG] 2025-01-09T07:21:39.775Z -         - Updated Spool-ID 9 => Bright Green
```

Rdiger-36, 2025-01-09:
Thats great, i also thank you for your effort to help me make the service a bit better 😃

You can now switch to the latest build, i released the new Update

closed 2025-01-09 by Rdiger-36