Rdiger-36/HaspelSync #13: Initial Setup - Error starting the service: - RequestError: Invalid URL

Roving-Ronin, 2025-01-18:
Hi,

Could anyone please assist, I'm trying to get this working as a docker on UNRAID and getting nowhere.

**Spoolman Docker**
I have Spoolman setup on the same host machine, and have tried it running on a Bridged network (so access via tcp/7912 that forwards to the docker tcp/8000) - VM subnet 10.0.1.x/24 - Access to Spoolman webgui works fine.

In case the Filament Status docker can't talk to it on the bridge or across VLAN / subnets (all other apps work fine cross subnets) I've also changed it to the Docker Hosts subnet 10.0.0.102 -  Access to Spoolman webgui works fine.

***bambulab-ams-spoolman-filamentstatus  Docker***
I've started with the 'latest' image and also placed it upon the default bridge for the VM subnet 10.0.1.x/24 and it starts and the GUI on tcp/4000 can be accessed. However it always fails with "Error starting the service: - RequestError: Invalid URL" and the Spoolman Connection Status and MQTT connection to printer always remain as "Disconnected"

In case there's some issue talking across the subnet / vlan I've also changed the docker to use the Docker Hosts subnet 10.0.0.101 and whilst I can access the GUI is still shows the same  "Error starting the service: - RequestError: Invalid URL" and the Spoolman Connection Status and MQTT connection to printer always remain as "Disconnected" errors.

This is with devices on:
- Printer: 10.0.0.100
- Filament Docker: 10.0.0.101
- Spoolman Docker: 10.0.0.102

I know that MQTT works normally, as I also have a BL LED connecting via MQTT usually. In case there's some conflict for access (I assume X1C and its MQTT can support multiple connections though ?) I've also disconnected the BL LED device.. still I can't get a connection.

Here's the Printer / Spoolman details used in 'latest'

[image]

I've also updated and tried the DEV build and used the printers.json, it starts with:
```
Printers loaded successfully: [
  {
    name: 'Bambu Labs X1C',
    id: '00M09D481300xxx',
    code: 'xxxxxxxx',
    ip: '10.0.0.100'
  }
]
[LOG] 2025-01-18T04:36:38.891Z - Setting up configuration... - 
[ERROR] 2025-01-18T04:36:38.896Z - Error starting the service: - RequestError: Invalid URL
```

Totally at a loss as to why this isn't connecting to the X1C printer (with 2 x AMS and all on latest firmware) or the Spoolman instance...?

Thank you.

Rdiger-36, 2025-01-18:
Hi, first of all thank you for supporting me. Your issue is very good structured and got the infos I need.

Because you use a dev version it may not be free from bugs.

I think I have also faced the same issue for making all ready for the new release and I might have fixed it.
 
Can you please try it with the official latest release?
It's not really necessary to use the dev channel when you only want to monitor and manage 1 printer.

Roving-Ronin, 2025-01-18:
@Rdiger-36 Thanks for taking a look. If its of assistance, here's the Docker Run for this docker:

```
docker run
  -d
  --name='Bambu-AMS-Filament'
  --net='br0'
  --ip='10.0.0.101'
  --pids-limit 2048
  -e TZ="Australia/Sydney"
  -e HOST_OS="Unraid"
  -e HOST_CONTAINERNAME="Bambu-AMS-Filament"
  -e 'PRINTER_ID'='00M09D481300xxx'
  -e 'PRINTER_CODE'='xxxxxxxx'
  -e 'PRINTER_IP'='10.0.0.100'
  -e 'SPOOLMAN_IP'='10.0.0.102'
  -e ' SPOOLMAN_PORT'='8000'
  -e 'UPDATE_INTERVAL'='15000'
  -e 'MODE'='automatic'
  -e 'SHOW_LOGS_WEB'='true'
  -e 'DEBUG'='true'
  -e 'TCP_PORT_4000'='4000'
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.webui='http://[IP]:[PORT:4000]'
  -l net.unraid.docker.icon='https://raw.githubusercontent.com/unraid/docker-templates/master/images/CA.png'
  -v '/mnt/user/appdata/spoolman-bambu-ams':'/app/printers':'rw'
  -it
  --restart unless-stopped 'ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:dev'
```

and the Spoolman docker:

```
docker run
  -d
  --name='Spoolman'
  --net='br0'
  --ip='10.0.0.102'
  --pids-limit 2048
  -e TZ="Australia/Sydney"
  -e HOST_OS="Unraid"
  -e HOST_CONTAINERNAME="Spoolman"
  -e 'PUID'='99'
  -e 'PGID'='100'
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.webui='http://[IP]:[PORT:8000]'
  -l net.unraid.docker.icon='https://raw.githubusercontent.com/m-klecka/unraid-templates/main/spoolman/spoolman.png'
  -v '/mnt/user/appdata/spoolman':'/home/app/.local/share/spoolman':'rw' 'ghcr.io/donkie/spoolman:latest'
```

As OP, it all looks good to me with Printer 10.0.0.100 / Filament Status Docker 10.0.0.101 Port 4000 / Spoolman 10.0.0.102 Port 8000 and serial number / code confirmed correct. Just not connecting to anything though.

Rdiger-36, 2025-01-18:
At the first look it must be all setup ok.
In your docket run is a space between the question mark and the S for SPOOLMAN_PORT.
Please check if that might be the fix if it's not a typo only for the issue.

Please check that if your container runs you can connect to Spoolman from the inside the container. Please also use the ENVs so you can check they are not the problem. You can use following commands:

``` bash
docker exec -it Bambu-AMS-Filament /bin/sh

wget -qO- http://${SPOOLMAN_IP}:${SPOOLMAN_PORT}/api/v1/health

```

Rdiger-36, 2025-01-18:
I could recreate you error by running your docker run command with the space between the quotes and the ENV --> ' SPOOLMAN_PORT'

Roving-Ronin, 2025-01-18:
Niklas, 
Profuse apologies for taking your time, I copied and pasted the key/variable names from your github page and looked all OK.
Just looked again and there was a space in front of 'SPOOLMAN_PORT'. Noob error = ~4 hours wasted. D'Oh.

Many thanks and apologies again.

Rdiger-36, 2025-01-18:
Hi @Roving-Ronin no Problem, everybody can make a mistake.
Sometimes it's the little things that are so simple that you overlook them.
Hope it works all fine now and this Service will help you

Roving-Ronin, 2025-01-18:
@Rdiger-36 This is interesting...

AMS1 has:
- Slot 0 - Bambu Lab PLA Black
- Slot 1 - Bambu Lab PLA Jade White
- Slot 2 - Bambu Lab PLA Bambu Green
- Slot 3 - Bambu Lab PLA Brown

AMS2 has:
- Slot 0 - Bambu Lab Support for PLA/PETG ( https://us.store.bambulab.com/collections/support/products/support-for-pla-petg )
- Slot 1 - Bambu Lab PLA Grey
- Slot 2 - Siddament PLA Translucent Green ( https://siddament.com.au/products/translucent-green-pla )
- Slot 3 - Bambu Lab PETG-HF Grey

Looking at your tool its showing:

[image]

BTW if/when do 'Actions' show as available?

Looking at Spoolman, it is showing the following:

[image]

So it looks like the AMS 2:
- Slot 0 with 'Support for PLA/PETG' isn't being imported into Spoolman.
- Slot 2 with 'Siddament PLA Translucent Green' also isn't being imported into Spoolman.

Would this be because SpoolmanDB doesn't appear to have an entry for 'Support for PLA/PETG' in https://github.com/Donkie/SpoolmanDB/blob/main/filaments/bambulab.json or for the 'Siddament PLA Translucent Green' in https://github.com/Donkie/SpoolmanDB/blob/main/filaments/siddament.json ?

As the data for Siddament filaments ( https://siddament.com.au/pages/slicer-profiles ) has all been imported into the Bambu Labs slicer (that is controlling the X1C / AMS's) I would have though that your dock would also detect/report this as 'Material = PLA Translucent' instead of <blank> for Material?

Rdiger-36, 2025-01-18:
Yes you mentioned it correctly, SpoolmanDB doesn't have entires for BambuLab Support Filament.

For your second spool problem, it's because my service doesn't support 3rd party spools, only Bambu Lab official spools with rfid chip

Roving-Ronin, 2025-01-18:
IIRC someone has already cracked the RFID codes from Bambu, sounds like we just need someone to work out how we can encode new ones and stick into 3rd party filament spools (and the multitude of colours some offer, that Bambu doesn't) so that the printers think they are seeing Bambu spools. :

Looking at the logs, it looks like currently Bambu might even have it coded to say 'if this isn't Bambu don't bother to track the %'. 

[B2]  B1FE98FF (0%) [[ 00000000000000000000000000000000 ]]

Many thanks for all your help ! 👍

closed 2025-01-18 by Roving-Ronin

Rdiger-36, 2025-01-18:
> IIRC someone has already cracked the RFID codes from Bambu, sounds like we just need someone to work out how we can encode new ones and stick into 3rd party filament spools (and the multitude of [...]

Yes i hope this will work someday in the future.

I also have support filament from Bambu, but until now I didn't got time to make a pull request in the SpoolmanDB. If you want, you can do it by your own. Donkie from Spoolman has written a manual how you can do this

Roving-Ronin, 2025-01-18:
SpoolmanDB, add Bambu Lab:

    Support for PLA/PETG
    Support for PLA
    Support for ABS
    Support for PA/PET

PULL Request https://github.com/Donkie/SpoolmanDB/pull/93

👍

Roving-Ronin, 2025-01-19:
@Rdiger-36 Niklas,  just stumbled upon this, thought it might be of interest to you. Was debating opening this as a new 'issue', not to fix anything but as a placeholder for storing this info / links to how to setup an OpenSpool and (potentially) down the track seeing if support for this could also be added, so Bambu users could automatically import both Bambu and non-Bambu spool data into Spoolman.

https://www.hackster.io/news/openspool-makes-proprietary-filament-a-thing-of-the-past-20cf8e85cef0

https://github.com/spuder/OpenSpool

The current version has the 'box beside the printer' ala the 'Mini', but appears there's also a PRO version (no info) and there a link to an AMS version and there's also a video where they've got it working with the AMS + spool holder on the side and reports Bambu + Non-Bambu spools into a X1.

https://openspool.io/hardware/openspool-ams.html

Rdiger-36, 2025-01-19:
Yeah that looks promising when there is an AMS support.
But I think that's not going in the right direction. 

First and foremost, the app is for tracking the filament that's not really working with no Bambu filament. Creating filament and spools in Spoolman is an side feature.

I will not say that at some time it doesn't will be supported, but it has no really significant focus for me.
Maybe if a have enough time I could imagine to give it a try.
But first of all OpenSpool have to support their AMS integration