Rdiger-36/HaspelSync #14: Docker Time Incorrect (TZData installed?)

Roving-Ronin, 2025-01-18:
Hi Niklas,

Just looking at the logs of the docker (BusyBox?) and noticed that all the time references are out by -11 hours.
Even though the docker run has TZ="Australia/Sydney" (which is GMT+11 at the moment due to daylight savings. Usually GMT+10) it appears that this Variable is ignored.

Just wondering if the docker build you have includes the tzdata package, that would then allow this variable to work?

```
apt-get update && apt-get install -y --no-install-recommends tzdata
```

Thanks.

Rdiger-36, 2025-01-18:
Hi, yes I'm already implemented it on my testing environment, but it's not released yet.
My plan is that I will release the official new version, with all these new and improved features you already know from the dev build, in the next 2 days.

Rdiger-36, 2025-01-19:
Hi, i have released the official latest release. Can you check if this fixed it for you?

Roving-Ronin, 2025-01-20:
The timezone data is being correctly applied from the host with this build.

Just switched to X1Plus firmware also (given BL's recent FU announcement re locking firmware / API in new firmware coming in 2-3 days time) and noticing that whilst I have AMS1 Full with BL filament and AMS2 Full (BL PLA Support in slot 0 / 3rd party in slot 2) otherwise BL filament, that AMS2 Slot 0 (BL PLA Support) and Slot 2 (Siddament Translucent PLA) are NOT showing.

Previously this was showing:

[image]

Now showing (B0 / B2 missing)
[image]

Logs:
[image]

Would you have a build number etc that I can use to rollback to the previous build and see if its an issue here of X1Plus ?
i.e. ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:**latest**

Roving-Ronin, 2025-01-20:
Based on your commit history I took a guess and found an image from 5 days ago, using:

https://github.com/rdiger-36/bambulab-ams-spoolman-filamentstatus/pkgs/container/bambulab-ams-spoolman-filamentstatus/337106732   (so pulled from ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus@sha256:cd459583f6e344ce56435522bc93b5e808089e2cf55752314356aeb2b275ec5e )

and that is working fine, showing all the installed filaments:

[image]

Log:
[image]

From this I'm guessing the latest build might have added a new bug ?

Rdiger-36, 2025-01-20:
Hi, no the Problem of the the old build was that it wants to process invalid data (no filament from Spoolman found for support and a 3rd party Spool). That's why it not detect it.
I want to implement a feature where these data will be shown as invalid and not be processed.

Roving-Ronin, 2025-01-20:
Knowing / seeing that there's a spool (even if not ID'ed - like B2 above) is better than if not showing anything.
Also B0 that is an official BL Support for PLA is showing an track, in latest build its shown as empty.

IMHO leaving it how you had it is better, so users can dive down to the AMS Dashboard as a cross check, but the 'primary' application being used is Spoolman, that doesn't show it unless its ID'ed in the SpoolmanDB filament.json
At least that way your tool is showing the 'actual' spools, but Spoolman is only showing and cataloging / tracking those that are ID'ed.

?

Rdiger-36, 2025-01-20:
Yes I know what you mean and as I already said I will bring these back, but I have to change a bit for the processing of the invalid or not in Spoolman recognized spools.
At first I thought this would be a better solution, but you are right and I changed my mind.
I will tell you when it's back

Rdiger-36, 2025-01-21:
Hi, that's weird I will test it with my own Support Spool from Bambu.

I also have tested a new logic that always shows all AMS Slots, even if there is no Spool loaded.
It could looks like this, but I'm not quite finished yet:

[image]

Rdiger-36, 2025-01-21:
Just found out that the external filament, which I use to create a new one, has for all filaments the same structure:
```bash
"id": "bambulab_pet-cf_black_500_175_n",
```

For the supports there is these structure:
```bash
"id": "bambulab_pla_supportforplablack_500_175_n",
```

I have implemented this for support filaments.

Rdiger-36, 2025-01-21:
This is from the api http://spoolman-ip:spoolman-port/api/v1/external/filaments

I'm still working on it but if you want you can check the dev build.

I'm also working on tags, so ist easier to go back.

Roving-Ronin, 2025-01-21:
I'll wait for the next build...2:22am time to Zzzzz

Rdiger-36, 2025-01-22:
The latest version is now released. Hope it contains all what you need

Roving-Ronin, 2025-01-22:
Assume its on DEV (Latest drops the emply / Support for PLA / 3rd party spools). Using Dev I get:

[image]

Thanks !

Rdiger-36, 2025-01-23:
I also have the problem von latest. I doen't know why, because it's the same.
I will check it and release it when it's fixed

Roving-Ronin, 2025-01-24:
Nothing wrong with my Spoolmam it was the latest version of Filament Status docker, that dropped showing the 'Support for PLA' and 3rd party spool taking a space in the AMS. Has 'latest' been updated and resolves this ?

Rdiger-36, 2025-01-24:
Can you please try to pull the latest image manual?
Mine works good even if I have loaded a Bambu, 3rd party or no spool

Roving-Ronin, 2025-01-25:
Just updated from Dev to Latest.

[image]

Pushed to Spoolman:

[image]

Rdiger-36, 2025-01-25:
Mh that's all a bit weird.
Do you have some logs?
I really want to make it working for you 🤔

Roving-Ronin, 2025-01-25:
```
[LOG] 2025-01-25_23:14:50 - Bambu Labs X1C -         - Updated Spool-ID 5 => Gray
[LOG] 2025-01-25_23:14:50 - Bambu Labs X1C -     - [B2] N/A B1FE98FF (0%) [[ N/A ]]
[LOG] 2025-01-25_23:14:50 - Bambu Labs X1C -     - [B3] PETG HF ADB1B2FF (19%) [[ 612E587AD2E4490082AA4D5EEB3AA66C ]]
[LOG] 2025-01-25_23:14:50 - Bambu Labs X1C -         - Updated Spool-ID 6 => HF Gray
[LOG] 2025-01-25_23:14:50 - Bambu Labs X1C - 
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C - AMS [A] (hum: 3, temp: 27.7ºC)
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -     - [A0] PLA Basic 000000FF (67%) [[ 881FBE543F824DBF9CF2528D04627578 ]]
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -         - Updated Spool-ID 1 => Black
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -     - [A1] PLA Basic FFFFFFFF (100%) [[ 5B82553133034AB38492D00B64090C28 ]]
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -         - Updated Spool-ID 2 => Jade White
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -     - [A2] PLA Basic 00AE42FF (16%) [[ 5C956947123543F0BD809F9219E23B9C ]]
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -         - Updated Spool-ID 3 => Bambu Green
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -     - [A3] PLA Basic 9D432CFF (100%) [[ E12B1A51918E444D852450AD5334E83B ]]
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -         - Updated Spool-ID 4 => Brown
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -     - [B0] Support for PLA 00000000 (41%) [[ E4EECF51B93D4269B965FE77C229984B ]]
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -         - Updated Spool-ID 7 => Support for PLA Black
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -     - [B1] PLA Basic 8E9089FF (79%) [[ 153EE35D6D874DAF94FB782293061FF9 ]]
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -         - Updated Spool-ID 5 => Gray
[LOG] 2025-01-25_23:16:50 - Bambu Labs X1C -     - [B2] N/A B1FE98FF (0%) [[ N/A ]]
[LOG] 2025-01-25_23:16:51 - Bambu Labs X1C -     - [B3] PETG HF ADB1B2FF (19%) [[ 612E587AD2E4490082AA4D5EEB3AA66C ]]
[LOG] 2025-01-25_23:16:51 - Bambu Labs X1C -         - Updated Spool-ID 6 => HF Gray
[LOG] 2025-01-25_23:16:51 - Bambu Labs X1C - 
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C - AMS [A] (hum: 3, temp: 27.7ºC)
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -     - [A0] PLA Basic 000000FF (67%) [[ 881FBE543F824DBF9CF2528D04627578 ]]
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -         - Updated Spool-ID 1 => Black
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -     - [A1] PLA Basic FFFFFFFF (100%) [[ 5B82553133034AB38492D00B64090C28 ]]
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -         - Updated Spool-ID 2 => Jade White
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -     - [A2] PLA Basic 00AE42FF (16%) [[ 5C956947123543F0BD809F9219E23B9C ]]
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -         - Updated Spool-ID 3 => Bambu Green
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -     - [A3] PLA Basic 9D432CFF (100%) [[ E12B1A51918E444D852450AD5334E83B ]]
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -         - Updated Spool-ID 4 => Brown
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -     - [B0] Support for PLA 00000000 (41%) [[ E4EECF51B93D4269B965FE77C229984B ]]
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -         - Updated Spool-ID 7 => Support for PLA Black
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -     - [B1] PLA Basic 8E9089FF (79%) [[ 153EE35D6D874DAF94FB782293061FF9 ]]
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -         - Updated Spool-ID 5 => Gray
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -     - [B2] N/A B1FE98FF (0%) [[ N/A ]]
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -     - [B3] PETG HF ADB1B2FF (19%) [[ 612E587AD2E4490082AA4D5EEB3AA66C ]]
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C -         - Updated Spool-ID 6 => HF Gray
[LOG] 2025-01-25_23:18:51 - Bambu Labs X1C - 
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C - AMS [A] (hum: 3, temp: 27.7ºC)
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -     - [A0] PLA Basic 000000FF (67%) [[ 881FBE543F824DBF9CF2528D04627578 ]]
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -         - Updated Spool-ID 1 => Black
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -     - [A1] PLA Basic FFFFFFFF (100%) [[ 5B82553133034AB38492D00B64090C28 ]]
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -         - Updated Spool-ID 2 => Jade White
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -     - [A2] PLA Basic 00AE42FF (16%) [[ 5C956947123543F0BD809F9219E23B9C ]]
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -         - Updated Spool-ID 3 => Bambu Green
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -     - [A3] PLA Basic 9D432CFF (100%) [[ E12B1A51918E444D852450AD5334E83B ]]
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -         - Updated Spool-ID 4 => Brown
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -     - [B0] Support for PLA 00000000 (41%) [[ E4EECF51B93D4269B965FE77C229984B ]]
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -         - Updated Spool-ID 7 => Support for PLA Black
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -     - [B1] PLA Basic 8E9089FF (79%) [[ 153EE35D6D874DAF94FB782293061FF9 ]]
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -         - Updated Spool-ID 5 => Gray
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -     - [B2] N/A B1FE98FF (0%) [[ N/A ]]
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -     - [B3] PETG HF ADB1B2FF (19%) [[ 612E587AD2E4490082AA4D5EEB3AA66C ]]
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C -         - Updated Spool-ID 6 => HF Gray
[LOG] 2025-01-25_23:20:52 - Bambu Labs X1C - 
[LOG] 2025-01-25_23:22:53 - Bambu Labs X1C - AMS [A] (hum: 3, temp: 27.7ºC)
[... cut ...]
[LOG] 2025-01-25_23:22:53 - Bambu Labs X1C -     - [B1] PLA Basic 8E9089FF (79%) [[ 153EE35D6D874DAF94FB782293061FF9 ]]
[LOG] 2025-01-25_23:22:53 - Bambu Labs X1C -         - Updated Spool-ID 5 => Gray
[LOG] 2025-01-25_23:22:53 - Bambu Labs X1C -     - [B2] N/A B1FE98FF (0%) [[ N/A ]]
[LOG] 2025-01-25_23:22:53 - Bambu Labs X1C -     - [B3] PETG HF ADB1B2FF (19%) [[ 612E587AD2E4490082AA4D5EEB3AA66C ]]
[LOG] 2025-01-25_23:22:53 - Bambu Labs X1C -         - Updated Spool-ID 6 => HF Gray
```

How should it be looking in both apps ?

Rdiger-36, 2025-01-25:
For me, it looks all good. Is this from the latest build?

The remaining percentage and the spool data is right collected.

Is the problem with 0% remaining still there?

Roving-Ronin, 2025-01-25:
Same 'latest' build as per the screen shots I uploaded yesterday.

Rdiger-36, 2025-01-25:
Ok, can you test it with debug logs and send me them?

Rdiger-36, 2025-01-25:
Can you also enter the dev mode on your browser?
If there's a problem with the frontend, you can see these errors there

Roving-Ronin, 2025-01-25:
The 0g / 0% is because its a non Bambu Labs roll / no RFID, so that is correct.

I do note that your non Bambu spool (4 days ago) for this in the serial number column reported "Non Bambu Lab Spool (No Serial)" , mine just says "N/A".

Ahhh just realised, looking at old screen grab above, that all my rolls where showing 0g / 0%. I think that must have been as the X1C/AMS was only turned on a few minutes before (doing work on the smart power monitoring plugs and has to unplug for a while to get them to boot new firmware... crap plugs from Arlec grrrr) and/or might need to do a job on X1C to get data showing.

Here's a new screenshot with 'latest', with the printer being on for a number of hours / having done a print:

[image]

The one thing that is wrong though is the colour of the 'Support for PLA/PETG'. Checking my records and I ordered the 'Nature' version and received that (that on https://au.store.bambulab.com/products/support-for-pla-petg?id=44722896208035 is shown as being available in "Nature" [being a semi translucent white] and Black).

Yet if you look at this time half way down the page, it shows 'Support for PLA/PETG' with only 'Nature' as the colour. 

[image]

The 'Support for PLA' though shows White and Black as available colours, my understanding is that this product has been superceded, as you can now only order 'Support for PLA/PETG' on the BL Store.

This is where is gets confusing... based on the BL data (above) I added these into the SpoolmanDB at: 
- Support for PLA/PETG: https://github.com/Donkie/SpoolmanDB/blob/3656107d3d0d6a03e075ce0d9ee7003ce1b23f11/filaments/bambulab.json#L767
- Support for PLA: https://github.com/Donkie/SpoolmanDB/blob/3656107d3d0d6a03e075ce0d9ee7003ce1b23f11/filaments/bambulab.json#L793 

Given I've entered the correct info into SpoolmanDB, why is it showing this in the logs though?:

~~~
[LOG] 2025-01-26_01:57:26 - Bambu Labs X1C - - [B0] Support for PLA 00000000 (41%) [[ E4EECF51B93D4269B965FE77C229984B ]]
[LOG] 2025-01-26_01:57:26 - Bambu Labs X1C - - Updated Spool-ID 7 => Support for PLA Black
~~~

So its being reported as 'Support for PLA' and the Black version, when the spool is actually a 'Support for PLA/PETG" in 'Nature' colour ?

[image]

You probably know more about how/what the RFID tags report... Any idea why this roll is incorrectly reporting itself ? Or how the Spoolman filament data is matched in your docker and convertered to that name in the logs?

Would you have a roll of the old 'Support for PLA' or the current 'Support for PLA/PETG' that you could put in your AMS and see what it reports itself as ?  Still doesn't explain why it shows as colour = Black... as both of these types of support have White and Nature with (hex colour: FFFFFF) ?

Rdiger-36, 2025-01-25:
Ahhh, now I understand why there is 0g remaining.
All the time I thought you have this problem 😂

> I do note that your non Bambu spool (4 days ago) for this in the serial number column reported "Non Bambu Lab Spool (No Serial)" , mine just says "N/A".

That's because I thought that's not right to put these info in this cell, so I've created the new column "Slot state". This could prevent some failures. So in the end there is no SN and that's why I put N/A in the cell.

---

For the Support Filament problem:

The Support filament is a bit tricky to integrate. It's not easy to find the matching right filament from Spoolman, because Bambu Labs data is false.
Your logs show, that your AMS reads the infos from the RFID chip and get the color "000000000".
This hex color is defined as "black" and fully transparent. The first 6 numbers defines the color and the last 2 defines the transparency. If it's 00 than it's fully transparent, if it's FF than it's non transparent.
The AMS sends the wrong color code so my app process this data and get the color black.
The biggest problem is, that Bambu doesn't write the name in their tags as they shown on their website. You can see it when it's get detected as "Support for PLA".

I'm not the friend of hardcoding this data, because for every change I need to release a new update. So I want to recognize it by the given data and the app should do the rest to find the matching filament by this.

> Would you have a roll of the old 'Support for PLA' or the current 'Support for PLA/PETG' that you could put in your AMS and see what it reports itself as ? Still doesn't explain why it shows as colour = Black... as both of these types of support have White and Nature with (hex colour: FFFFFF) ?

Yes, I happen to have some here. Bambu gave it to me for free when I bought my printer. But it looks like it's the "old" white support and it's recognized right:
[image]

If you want, you can send me the raw mqtt data from your AMS. For this you can do this when your on Linux:

```bash

openssl genrsa -out ca.key 2048

openssl req -new -x509 -days 3650 -key ca.key -out blcert.pem

mosquitto_sub -h PRINTER_IP -p 8883 --cafile ./blcert.pem --insecure -u bblp -P PRINTER_ACCESCODE -t device/PRINTER_SERIAL/report

```

If you run a different OS, then tell me witch one you use and I will give you your needed commands

Roving-Ronin, 2025-01-26:
Not sure re the above commands, but I can see in my Home Assistant the following from the MQTT.

Here is the 2nd AMS (ala AMS1 as 0 is the first one. And this is the Slot 1...ala Tray 0):
i.e. This is the one with the BL 'Support for PLA/PETG' in 'Nature' (semi translucent White).

[image]

If of interest, here's the Siddament Translucent Green (no RFID), thats in the second AMS / Slot 3:

[image]

PS. The 'Tray UUID' is the serial number of the spool. If BL only lists 'Support for PLA' as an ID for both the PLA/PETG and the old PLA spools, with the old PLA no longer sold, it does raise the issue is dropping the old PLA from SpoolmanDB the best option. At least this was, unless there's some other way you can ID / match it (?), at least that would be a long term solution to have the product people can buy and use in future be correctly displayed, versus getting the ongoing product incorrect and the old one that will fade away correct.

???

Rdiger-36, 2025-01-26:
Thank you for providing me with your data. This data from HomeAssistant is also OK, it is the same data that you receive with the commands I mentioned above.

It's like I said earlier, that the spool contains false data. For this I couldn't do anything for a right automation. But you can manually create a filament and spool and add the serial (tray_uuid) in your created spool from Spoolman. 

In the current version, the data to be displayed is loaded from the external database. It's easier to implement this method for all functions. But I can add a new logic, that my app could use the data from the already connected spool from Spoolman.
This would fix your problem.

So I think the best solution is to do both: drop the old Support filaments from SpoolmanDB and fetch the displayable data from already existing and connected spools

Roving-Ronin, 2025-01-26:
its still weird thought that the Support for PLA/PETG RFID is reporting colour = 0000000 (Black) when it should be reporting FFFFFFFF (White).

Here's the current code for the SpoolmanDB, showing correct hex codes. Left wondering if BL (or their outsourced filament making supplier) has stuffed up and is sticking Black RFID codes on Nature spools and vice versa. Wondering if this should be swapped around for SpoolmanDB (and if they fix up the RFID tags in future, it can be swapped back) ?

```
        {
            "name": "Support for PLA/PETG {color_name}",
            "material": "PLA",
            "density": 1.28,
            "weights": [
                {
                    "weight": 500,
                    "spool_weight": 250
                }
            ],
            "diameters": [
                1.75
            ],
            "extruder_temp": 210,
            "bed_temp": 55,
            "colors": [
                {
                    "name": "Nature",
                    "hex": "FFFFFF"
                },
                {
                    "name": "Black",
                    "hex": "000000"
                }
            ]
        },
```

Roving-Ronin, 2025-01-26:
OK let me know re this 'customized code'.... assuming your saying you can take hex = black in and output as hex = white?

Rdiger-36, 2025-01-26:
Yes I tested it and you only have to manually edit the name and the color of your filament:

[image]

[image]

After Edit:

[image]

[image]

I will also cut of these "Support for..." and other things

Roving-Ronin, 2025-01-26:
I see how to manually edit it on the Filaments tab of Spoolman, but how did you get it to update that into your AMS Status tool?

I changed mine in Spoolman and it still shows as 'Support for PLA Black' (with black background colour) in AMS Status tool.

Rdiger-36, 2025-01-26:
It's one little test from my private development. I'm still working on the code to make it more stable. I will tell you when you can test the new dev build

Rdiger-36, 2025-01-26:
I released the new dev build. You can test it if you want

Roving-Ronin, 2025-01-27:
Installed DEV and updates with:

[image]

Whilst 'Support for PLA' (actually PLA/PETG) now shows White (correct) instead of Black, I also noticed that the B2 slot, containing a Siddaments Translucent Green PLA spool and that was showing:

[image]

is now show it as BL:

[image]

Rdiger-36, 2025-01-27:
Ok, that looks good. This little mistake will be corrected. I forgot to adjust one function.

I will tell you when it is fixed

Rdiger-36, 2025-01-27:
The problems should be solved in the newest dev build

Rdiger-36, 2025-01-28:
After some testing, I couldn't find a problem with this. It's now implemented in latest build I released now

Rdiger-36, 2025-01-29:
No, but I have a list to cut them and other unwanted things out, but it's not complete yet

Rdiger-36, 2025-02-01:
I have implemented a solution to cut off these materials in the colour filed. However, this only applies to the name of the colour in the colour field on the website. In Spoolman the value is set from the external Spoolman database, this has the suffix HF, Support For, Silk etc.

I will test it in the new dev build and release it when it all works fine

Rdiger-36, 2025-02-03:
It's now integrated in the latest dev build. After some more testing I will release it