Rdiger-36/HaspelSync #15: [Bug] Bambu Studio and MQTT crash

ados8, 2025-01-19:
I'm not sure where issues should go, here or at the OpenSpoolman repo.
In any case, I've just set it up and noticed if Bambu Studio (BS) crashes/looses connection if the docker is running.
In some cases the BS app crashes or in most cases it looses the printer and cannot connect.

[image]

OpenSpoolman will also stop reporting any filaments in the AMS.
This happens the moment the devices tab is selected with the printer which is required for LAN mode.
As soon as the docker is shutdown the printer connects and doesn't ask for the access code.
I assume just because there's 2+ connections polling at the same shouldn't be an issue.

OpenSpoolman logs show it doesn't connect to the MQTT connection until the moment you click the devices tab.

[image]

Only then does it populate in the logs and the web browser.
This is really a minor issue, might even be the limits of AMS?
Either way it only crashes very rarely and the benefits' are completely worth stopping the docker, connection in BS and then starting docker.

Also noticed the wood filament doesn't have any actions, this is a new Spoolman instance.
I fixed it by manually searching and adding the filament which for 3F231CFF is Bambu Lab - Rosewood.
This seems to be an RFID issue, in BS it's listed as just Bambu PLA Wood.

[image]

ados8, 2025-01-19:
MQTT lockout/crashing is more regular than I thought.
Was working and then changed the colors with crash of MQTT with BS, dropped the printer with error above.
Restarted the container and same issue, what works is having the container off and then reconnecting using the prompted access code.
Happening enough I have to keep the access code in my clipboard permanently.

Turned of debugging logs at the time of issue but nothing to see.
When the even occurred it was saying the following:
```
Waiting for MQTT messages...
- Parse message to processable JSON
- Check if message contains AMS Data
- No processable Data found for JSON filter data.printer.ams.ams
```
However, it says this during the times it is working and has nothing to sync.
Bambu Studio: 1.10.1.50
Bambu P1S: 01.06.01.02
AMS: 00.00.06.44

ados8, 2025-01-19:
100% confirmed is this docker and the logs provide no insight.
After it stopped again I tried close to 10 times in a row to connect with BS and it failed every time.
Stopped any other dockers I have using the printer and same issue.
Restarted the this docker and same issue.
Stopped the docker and BS connected on next attempt.

Rdiger-36, 2025-01-19:
Hi, thanks for your reply.
I will take a look at it.
Until now, I didn't got these connection issues when i was working with both Apps (BS and my service).
How many instances do you run, that connects to the MQTT system of the Bambu Lab Printer?

I use following instances simultaneously and got no issues:
- Bambu Studio
- Bambu Handy
- Home Assistant Integration
- my own docker container

ados8, 2025-01-19:
Created a new Linux docker container and same issue on same version, rules out Windows.
Run the following:

- Bambu Studio
- OctoEverywhere
- Home Assistant Integration
- this new docker

Stopping OE did nothing to resolve the issue.
I'll try going back to older BS versions and report back

ados8, 2025-01-19:
Umm, I really wanted to get this working and have been trying everything, I changed the image to dev and it works; well done you fixed it 😅.
I didn't change anything else, not using the JSON config, just the one printer.
I've tested many times in a row and nothing but the docker has changed and it works.
I've closed/opened BS many times, restarted the docker many times and even run another compose, solved.

Rdiger-36, 2025-01-19:
Ok, than there must be some really unique bugs, that not affected other users.
Its good that it is "solved" in the dev build., and we take a look at it when I release the dev build official to latest.
Until then, please let this issue open

Rdiger-36, 2025-01-19:
> Also noticed the wood filament doesn't have any actions, this is a new Spoolman instance.
I fixed it by manually searching and adding the filament which for 3F231CFF is Bambu Lab - Rosewood.
This seems to be an RFID issue, in BS it's listed as just Bambu PLA Wood.

This is weird, because on the Bambu Lab Store they define Rosewood with an other Color code
https://eu.store.bambulab.com/de/products/pla-wood?variant=53441583087964

Also in the SpoolmanDB there is the color Code from Bambu:
https://github.com/Donkie/SpoolmanDB/blob/main/filaments/bambulab.json

Rdiger-36, 2025-01-19:
The newest version is released, can you confirm that it also worked for you an d your issue?

ados8, 2025-01-20:
Sadly no it doesn't work, switched back to :dev and it works.
Had suspicions the image was not updating so removed and pull again, works.
Can I suggest adding to GitHub the release versions with release notes and tags, people generally use this to review changes and control the docker images they install.

Rdiger-36, 2025-01-20:
Ok, yes that’s what I wanted to add but I forgot it

closed 2025-01-21 by ados8