Rdiger-36/HaspelSync #16: Spools of the same color get merged even if they have a different tag

karaktaka, 2025-01-31:
Hey,

I think I might have encountered a bug. If I use the same material and color twice in the ams, there is only one of the spools created in spoolman and overwritten constantly by either one or the other. So sometimes the spool is already used or unused depending on which spool was send last to spoolman.

```
[LOG] 2025-01-31_21:00:40 - P1S - - [A0] PLA Matte FFFFFFFF (79%) [[ A1D3BA2D8F504723AAAE76A4288EA8B1 ]]
[LOG] 2025-01-31_21:00:40 - P1S - - Found mergeable Spool => Spoolman Spool ID: 38, Material: PLA Matte, Color: Matte Ivory White
[LOG] 2025-01-31_21:00:40 - P1S - merging Spool...
[LOG] 2025-01-31_21:00:40 - P1S - Spool successfully merged with Spool-ID 38 => Matte Ivory White
[LOG] 2025-01-31_21:00:40 - P1S - - [A1] PLA Matte FFFFFFFF (100%) [[ 48EF649D42C24312B0EAF10092C58F0D ]]
[LOG] 2025-01-31_21:00:40 - P1S - - Updated Spool-ID 38 => Matte Ivory White
```

You can see that both spools have a different uuid and based on your readme a spool with an existing tag should not be merged.

Rdiger-36, 2025-01-31:
Hi thank you for letting me know about this bug.

I will take a look at it.

Rdiger-36, 2025-02-01:
I've found the bug. I forgot a check in the merge function to only merge a spool if there is no tag. I will test it and release it in the dev build, if it works fine I will release it in the official latest build

Rdiger-36, 2025-02-03:
It's now integrated in the latest dev build. After some more testing I will release it

Rdiger-36, 2025-02-18:
It's now included in the new latest build 1.0.2

closed 2025-02-19 by Rdiger-36