Rdiger-36/HaspelSync #17: Cannot read properties of undefined (reading 'extra')

rotanon, 2025-02-01:
Containers starts running OK and updating spoolman as expected. 

After a period of time I start to see the following error. After this point, spoolman is no longer updated. If I stop and start the container, it starts to work again.

[ERROR] 2025-02-01_17:27:38 - Bambu Lab X1C - Error processing message for Printer: 00MXXXXXXXXXXXX - Cannot read properties of undefined (reading 'extra')

Snippet of the debug log attached for reference

[_bambulab-ams-spoolman-filamentstatus_logs.txt](https://github.com/user-attachments/files/18627646/_bambulab-ams-spoolman-filamentstatus_logs.txt)

Rdiger-36, 2025-02-01:
Hi thank you for letting me know about this bug.

I will take a look at it.

rotanon, 2025-02-01:
Awesome. Thank you. By the way, fantastic job on completing this integration. Was desperately needed.

Rdiger-36, 2025-02-03:
It's now integrated in the latest dev build. After some more testing I will release it

od3m, 2025-02-09:
Perfect. Thanks in advance. I have the exact same issue. 
Thank you for fixing it.

Rdiger-36, 2025-02-18:
It's now included in the new latest build 1.0.2

numindast, 2025-02-19:
Can confirm, the 1.0.2 build made this message go away for me as well.

closed 2025-02-19 by Rdiger-36