Rdiger-36/HaspelSync #18: Printer Not Conectet

Toby-ch, 2025-02-01:
Hello,

I have the problem that the container cannot receive MQTT data.
I performed a resync on the AMS and restarted the printer. In Home Assistant, the data is being recorded, but not in this container. However, the printer is reachable (both via LAN and Cloud).

[image]

Rdiger-36, 2025-02-01:
Hi, can you please run the container with the Environment variable "DEBUG=true"?

This could help me to locate the problem

Toby-ch, 2025-02-01:
Hi

I have added it, but there is no additional information except for:  
`[ERROR] 2025-02-01_21:41:50 - Server - Error starting the service: RequestError: connect ECONNREFUSED 10.98.4.27:8000`  

However, this only appears when the Spoolman container has not yet been started. If I then restart the FilamentStatus container afterwards, the message disappears.

[image]

Rdiger-36, 2025-02-01:
Ok thank you. 
Which Intervall have you set in you ENV?
Can you manually trigger a mqtt message?
You can do this by open the Bambu Handy app or load/unload a spool in your AMS

Toby-ch, 2025-02-02:
_Which Intervall have you set in you ENV?_

I think these are the default settings. Here is the container's configuration:

b```
ambulab-ams-spoolman-filamentstatus:
    image: ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:latest
    container_name: bambulab-ams-spoolman-filamentstatus
    ports:
      - 4000:4000
    environment:
      - SPOOLMAN_IP=10.98.4.27
      - SPOOLMAN_PORT=8000
      - UPDATE_INTERVAL=120000
      - MODE=automatic
      - DEBUG=true
    volumes:
      - /mnt/user/appdata/Spoolman/ams:/app/printers
    restart: unless-stopped
    networks:
      br0.11:
        ipv4_address: 10.98.11.218
```

I have configured the container in the same VLAN as the printer without success. The only achievement is that the web interface is now displayed on port 4000, which didn’t work before. However, MQTT still doesn’t work. I have reloaded the settings on the printer.

![Image](https://github.com/user-attachments/assets/82304b50-eab5-4df4-a2fc-53d4f1569642)

Ping works to all containers, the VLANs are open, and there are no blocking rules.

![Image](https://github.com/user-attachments/assets/d90f67d2-8e4e-4e96-b432-59b83370d6eb)

I checked the firewall log, and the printer tried to respond to the container while it still had the old IP address.

(Note: There is a 1-hour time difference between the firewall log and the container log!)

![Image](https://github.com/user-attachments/assets/b11a3146-088b-4ac7-b0d9-89eec2b58de5)

Rdiger-36, 2025-02-03:
Wir können auch ruhig auf deutsch weiter schreiben, ist glaube ich einfacher.

Ich arbeite gerade an einem Script, welches man im Container direkt aufrufen kann. Dort wird die Verbindung mit dem Drucker hergestellt und alle MQTT Nachrichten ausgegeben.
Damit kann man dann testen, ob die Kommunikation zwischen Container und Drucker soweit in Ordnung ist um dies erstmal als Fehlerquelle auszuschließen.

Ich bin aber noch nicht ganz fertig. Ich würde dir dann Bescheid geben

Toby-ch, 2025-02-03:
Hy
"Damit kann man dann testen, ob die Kommunikation zwischen Container und Drucker soweit in Ordnung ist um dies erstmal als Fehlerquelle auszuschließen."
Laut dem UDM log gab es mehrfa verbindungs versuche:
SRC=10.98.11.82 ( Drucker) > Container DST=10.98.4.29
Source Port 8883 auf DPT=59624

Ich habe sicherheitshalber den Drucker und den Container nun im selben Vlan da ich nicht 100% sicher bin ob MQTT mit dem Vlan klar kam. wenn ich dieses log sehe scheint es jedoch nicht am Vlan zu lioegen.

Gibt es irgend welche mindest vorausetzungen das es funktioniert ? Firmware etc.
weil ich bin anscheinder der einzige bei dem es nicht funktioniert.

Ich bin aber noch nicht ganz fertig. Ich würde dir dann Bescheid geben_
Kein stress!

Rdiger-36, 2025-02-03:
> Gibt es irgend welche mindest vorausetzungen das es funktioniert ? Firmware etc.
weil ich bin anscheinder der einzige bei dem es nicht funktioniert.

Mh ne das sollte alles klappen. Solange du einen P1 oder X1 Drucker hast, ein AMS angeschlossen hast.
Welche FW hat denn sein Drucker und welches Modell ist es?

Ich wüsste auf Anhieb nicht wieso es nicht klappt, da die Home Assistant Integration funktioniert.
Zum Thema VLAN: Solange alles gepingt werden kann ist ja eine Verbindung da. MQTT verwendet TCP/IP (genauso wie ping), daher sollte es auch zwischen VLANs funktionieren, also so lange der Drucker gepingt werden kann.

Was natürlich sein kann ist, dass wieso auch immer, die Daten gelesen werden können aber nicht verarbeitet werden. Dafür bastel ich gerade das vorher erwähnte script.

Du könntest aber mal testen, ob du MQTT Nachrichten auslesen kannst, dafür musst du wie folgt vorgehen:

Für Linux musst Du dir einfach die Pakete openssl, mosquitto und mosquitto-clients in der commandline installieren. 

Für Mac OS müsstest Du mal googeln, wahrscheinlich klappt das da über Homebrew. Das hab ich selbst noch nicht gemacht.

Dann musst du wie folgt vorgehen:
 
1.	Zertifikat vom Drucker extrahieren:

```
openssl s_client -connect PRINTER_IP:8883 -showcerts
```

2.	Kopiere das Zertifikat (zwischen -----BEGIN CERTIFICATE----- und -----END CERTIFICATE-----).

3.	Speichere es als Datei:

```
nano bambu_cert.pem
```

4.	Dann mit mosquitto_sub verwenden:

```
mosquitto_sub -h PRINTER_IP -p 8883 -u "bblp" -P "ACCESSCODE" -t "device/PRINTER_SERIAL/report" --cafile bambu_cert.pem -d
```

Toby-ch, 2025-02-03:
_Welche FW hat denn dein Drucker und welches Modell ist es?_
X1C
von Bambu Lab
Firmware: 01.08.02.00 
AMS
von Bambu Lab
Firmware: 00.00.06.44
Ausgelsen dur Homeassistant

Du könntest aber mal testen, ob du MQTT Nachrichten auslesen kannst, dafür musst du wie folgt vorgehen:
das versuche ich morgen muss mir zuerst ein Linux system instalieren

Rdiger-36, 2025-02-03:
> das versuche ich morgen muss mir zuerst ein Linux system instalieren

Nicht nötig, das neue dev build wird voraussichtlich nachher online gehen, dort wird das Script bereits enthalten sein.

Näheres gebe ich noch Bescheid

Rdiger-36, 2025-02-03:
Das Script ist nun im neusten dev build integriert.

Um es nutzen zu können logge dich im container ein: docker exec -it CONTAINER_NAME /bin/sh
Danach kannst du einfach "debug-printers" eingeben dir die MQTT Nachrichten anzeigen lassen bzw. evtl auftretende Fehler

Toby-ch, 2025-02-04:
Ich habe das Script einmal ausgeführt:

[image]

Schein kein Problem mit dem MQTT zu geben.

Rdiger-36, 2025-02-04:
Doch, an den MQTT messages liegt es. Es sollte ungefähr so aussehen:
[image]

Irgendwas in deinem Netzwerk scheint blockiert zu sein.
Kommt der Port 8883 bei deiner FW durch?

Toby-ch, 2025-02-04:
Hmm
Da sich der Container wie auch der Drucker im selben Vlan befinden geht nichts über die Firewall.

Rdiger-36, 2025-02-04:
Ach so stimmt, das hattest du ja vorher schon erwähnt. 
Ich prüfe mal ob es noch weitere Möglichkeiten gibt

Rdiger-36, 2025-02-04:
Ist das ein Fehler in der Anzeige oder beginnt deine Seriennummer mit zwei O (Buchstabe)?
Ich glaube das sollten 2 nullen sein --> 0

Probiere mal bitte damit aus

Rdiger-36, 2025-02-07:
War das der Fehler?
Wenn ja, dann würde ich den issue schließen

Toby-ch, 2025-02-07:
Hy ich hab ers jetzt deine Nachriucht gesehen Ja das war der Fehler Copy past error ich habe die SN aus dem Foto rauskopiert und O und 0 ist leider zum verwächslen ähnlich.

closed 2025-02-07 by Rdiger-36