Rdiger-36/HaspelSync #19: It is possible to use Bambulab HA sensor to update spools usage?

platini76, 2025-02-03:
I ask if it is possible to use the accurate sensor.a1_grammatura_stampa that is present in bambulab HA to update filament status usage.
thanks

Rdiger-36, 2025-02-03:
This could work and would even function without Home Assistant. The necessary data is provided directly by the printer. However, there is a major problem: this data does not indicate which filament, colors, or spools are being used. It only shows the total amount of filament required—without any additional details.

This makes it difficult to accurately determine the actual filament consumption and deduct it from the corresponding spool in Spoolman. While this might work for a single-color print, there are still further issues: in the event of a failed print or an aborted job, there is no precise information on how much filament has already been used.

platini76, 2025-02-03:
can be a solution for most problems to add a selector for the spool not recognized where you can select the actual spool from spoolman db?

Rdiger-36, 2025-02-03:
Yes this could be possible, but only for a manual system, not automated.
I thought about it and my idea how to implement it could be a new Menu where the last print is visible with all necessary data and you can select a spool from spoolman and click on a button like "calculate printed filament". This would subtract the printed/used filament from the spool from spoolman.

How ever this could take some time to implement it and it would interfere with the other system. Because if you load the spool later in your AMS, this calculated remaining filament would be overwritten on the estimate value from the AMS.

Im not 100% sure if it would be useful.
I think its easier to set the data manual in spoolman

platini76, 2025-02-03:
did you think to a pop-up on homeassistant tiles.. that says... hey your last print is not sync... would you like to sync?

Rdiger-36, 2025-02-03:
No, rather an implementation in my app.
But unfortunately I don't see the advantages of it.
If you want something like that in Home Assistant then you have to ask others, I'm the wrong person to talk to about integrations into Home Assistant. Sorry

platini76, 2025-02-03:
seems noone has interest in this.. :(

Rdiger-36, 2025-02-03:
I don't think so. The problem is that Bambu doesn't provide that much information to work with. You have to work very hard to make things like this work. For all these ideas, you have to see with a rational eye. If it is too complicated for a ‘little thing’, then it is not profitable and sometimes it is easier to do things manually, such as setting a new calculated value in Spoolman directly.

platini76, 2025-02-03:
have you seen this?
https://www.reddit.com/r/BambuLab/comments/1hamev5/keeping_spoolman_uptodate_with_bambu_lab_and_home/

Rdiger-36, 2025-02-03:
No, not yet. I know the creator of this integration from Spoolman issue 217, from which I also got the inspiration for this app. It is based on the idea of a contributor from Spoolman Diogo Resende. Anyone who wants to can build their own integration.

platini76, 2025-02-03:
it is a simple script... (for me not working but maybe it's my setup the error) but it can be simply readapted to use sensor of filament usage

closed 2025-02-04 by Rdiger-36