Rdiger-36/HaspelSync #21: Server - Failed to read log file for printerId "xxx": ENOENT: no such file or directory, open '/app/logs/01P09A451702885.log'

luis15pt, 2025-02-05:
Not sure if anyone ran into this issue but i fixed it by mounting the following volume  "/app/logs" to the container

  bambulab-ams-spoolman-filamentstatus:
    image: ghcr.io/rdiger-36/bambulab-ams-spoolman-filamentstatus:latest
    container_name: bambulab-ams-spoolman-filamentstatus
    ports:
     - 4000:4000
    environment:
      - SPOOLMAN_IP=192.168.8.250 #IP address of your Spoolman instance
      - SPOOLMAN_PORT=7912 #Port of your Spoolman instance
      - UPDATE_INTERVAL=120000 # Time in ms for updating spools in Spoolman (standard 120000 ms -> 2 minutes) min. 1000
      - MODE=automatic
    volumes:
      - /share/Container/spoolman:/app/printers
      - /share/Container/spoolman:/app/logs
    restart: unless-stopped

Rdiger-36, 2025-02-06:
Hi until now this problem didn't show up.

I will take a look at it, thank you for letting me know about this issue

Rdiger-36, 2025-02-18:
It should work now and it's now included in the new latest build 1.0.2

Rdiger-36, 2025-02-25:
Its ready for testing in the new dev build version 1.0.4

Rdiger-36, 2025-03-02:
Its finally released int he latest build version 1.0.4.
I will close this Issue