Rdiger-36/HaspelSync #22: Does not appear to be processing PETG HF Red filament rolls correctly

rotanon, 2025-02-08:
As per the attached screenshot, it doesnt seem to be able to resolve the PETG HF Red correct. 

Issue 1 - Spool is listed in bambulab-ams-spoolman-filamentstatus screen, but color  is only shown in the hex code. 
Issue 2 - This spool is not registered in spoolman app at all.

Attached screenshot of filament status, and debug log entries

[bambulab-ams-spoolman-filamentstatus.csv](https://github.com/user-attachments/files/18717277/bambulab-ams-spoolman-filamentstatus.csv)
[image]

Rdiger-36, 2025-02-08:
It seems that Bambu Lab have some inconsistent data. Your Spool got a other recognized Color Code than Bambu shows on their website:

https://eu.store.bambulab.com/products/petg-hf

This data is also integrated in Spoolman and its database, that's the problem.
It's the 2nd time this problem appear that this codes doesn't match to their data.
I think it's a good idea to implement a feature to customize/assign the filament over the web app so this data will be used later.

If you want you can add a new filament and spool manually in Spoolman and connect it with your serial number. After this the color should be recognized right.

I will add this customization/assignment later, because I'm I'll

ados8, 2025-02-21:
What about a notification system, webhooks?
This really is a BBL problem but it would be great to work around it best we can.
If the user adds a role and it doesn't exist then webhook sends notification and then user knows to manually add in Spoolman etc.

Rdiger-36, 2025-02-24:
Firstly I will implement it like this:
I am adding a new column called 'State' to indicate whether an error occurred while determining the filament. If an error is detected, a button will be used to display this information:

[image]

By clicking on "Show Info!" the following dialog window is opened, from which it is also possible to open Spoolman directly in the creation of a filament:

[image]

This can be tested in the new dev build which will be released soon

I will think about whether it's really necessary to implement a notification system or if this solution does the job

Rdiger-36, 2025-02-25:
Its ready for testing in the new dev build version 1.0.4

Rdiger-36, 2025-03-02:
Its finally released int he latest build version 1.0.4.
I will close this Issue

closed 2025-03-02 by Rdiger-36