Rdiger-36/HaspelSync #23: Support for Spoolman when running in subfolder

heathen711, 2025-02-18:
Appears that all of the queries to Spoolman assume it is running as the root of the server, mine is running under a subfolder alongside other services.

Could an optional environment var be added to support this?

e.g:
`http://${SPOOLMAN_IP}:${SPOOLMAN_PORT}/api/v1/spool`
to:
SPOOLMAND_FOLDER="/my-sub-folder"
`http://${SPOOLMAN_IP}:${SPOOLMAN_PORT}%{SPOOLMAND_FOLDER}/api/v1/spool`

heathen711, 2025-02-18:
BTW: Spoolman supports this by setting `SPOOLMAN_BASE_PATH=my-sub-folder` in it's env.

heathen711, 2025-02-18:
[spoolman_base_folder_support.patch](https://github.com/user-attachments/files/18855715/spoolman_base_folder_support.patch)

Some devs are sensitive to people opening PRs, so I made a patch instead. Let me know what you think.

This is running on my host now and works (aside from my other issues I opened).

Rdiger-36, 2025-02-19:
That's also a good idea. I didn't knew this and I will add it in the next dev build.
If you want, I can tell you when its ready to test

Rdiger-36, 2025-02-23:
Its implemented and will be available in the next dev build coming next week

Rdiger-36, 2025-02-25:
Its ready for testing in the new dev build version 1.0.4

heathen711, 2025-02-26:
Works.

heathen711, 2025-02-26:
Make sure you update the Readme.md for the new env (I see them in the change log so I knew what they were)

Rdiger-36, 2025-03-02:
Its finally released int he latest build version 1.0.4.
I will close this Issue

closed 2025-03-02 by Rdiger-36