Rdiger-36/HaspelSync #26: Silk recognition

heathen711, 2025-02-18:
From the Readme you need more info to help with multi-color silk filament.

I have rolls, what info do you need?

[image]

heathen711, 2025-02-20:
I assume this is what you need:
```
{
    "bed_temp": "35",
    "bed_temp_type": "1",
    "cali_idx": 9183,
    "cols": [
        "FF9425FF",
        "C16784FF"
    ],
    "ctype": 1,
    "drying_temp": "55",
    "drying_time": "8",
    "id": "0",
    "nozzle_temp_max": "230",
    "nozzle_temp_min": "210",
    "remain": 80,
    "tag_uid": "BAA8227400000100",
    "tray_color": "FF9425FF",
    "tray_diameter": "1.75",
    "tray_id_name": "A05-T1",
    "tray_info_idx": "GFA05",
    "tray_sub_brands": "PLA Silk",
    "tray_type": "PLA",
    "tray_uuid": "0417584A3ABE4274838571DB6AA6CABA",
    "tray_weight": "1000",
    "xcam_info": "AC0D88132003E803CDCC4C3F"
},
```

I attached a full output for you too:
[mqtt.json](https://github.com/user-attachments/files/18892183/mqtt.json)

Rdiger-36, 2025-02-20:
Thank you very much!
I will take a look at it and see how I can implement it

Rdiger-36, 2025-02-23:
With your information I found a way to implement a double color feature. Its not ready at all but It will looks like this:

[image]

And yes, there was again the problem with the color codes...

Rdiger-36, 2025-02-25:
Its ready for testing in the new dev build version 1.0.4

heathen711, 2025-02-26:
I did not get the colors like the screenshot you shared:

[image]

Possibly related to the above but:
```
No machting Filament found in Database, please check manually!

This error shows up when the official data from BambuLab does not matches with the collected data from the spool!
```

Wouldn't you also want users to tell you so patches could be made?
This would hold true for every time I add a new color combo spool, then why not just have it made?

Rdiger-36, 2025-02-26:
How did you mean this?

Rdiger-36, 2025-03-01:
> 1. the PLA Silk dual combo color is not displaying correctly

Yes, because there’s false data on the RFID chip (the second colour code is incorrect). If you manually create the filament and spool, and enter your serial number in the “tag” field, it will be correctly identified because it will use the data from the Spoolman spool.

> 2. When I click on the show info, it tells me that the info from the spool does not match the info from BL, would that be a code issue or an issue with the spool (assuming that BL info is always correct)
note: That is the exact same spool that I send you the MQTT dump from, so it should display exactly like your screenshot if working...

The inaccurate data provided by the spool is the cause of this issue. Spoolman, which obtains its data from BambuLab, couldn’t find the matching filament. The database on Spoolman contains information from BambuLab’s store. For some reason, the RFID chips may have different colour codes than those listed on the website

heathen711, 2025-03-02:
Ahh... that's not confusing at all! Thank you for explaining.

Rdiger-36, 2025-03-02:
Yes, it also annoys me that they give different information on their website and write different data in the RFID chips. But I believe that this is a problem with their suppliers.

Rdiger-36, 2025-03-02:
Its finally released int he latest build version 1.0.4.
I will close this Issue

closed 2025-03-02 by Rdiger-36