Rdiger-36/HaspelSync #28: [Optional] Frontend.js support for relative urls

heathen711, 2025-02-19:
Like my spoolman instance, I'm hosting this instance and proxy-ing it to a subfolder.

I made adjustments to the frontend.js to use relative pathing to support this, the alternative is to have it know the frontend path in some config.

I'll let you decide if you want this.

[frontend_relative_pathing.patch](https://github.com/user-attachments/files/18856049/frontend_relative_pathing.patch)

Rdiger-36, 2025-02-19:
Hi thx for your request. That's a good idea. I didn't knew this and I will add it in the next dev build.
If you want, I can tell you when its ready to test

Rdiger-36, 2025-02-23:
Its implemented and will be available in the next dev build coming next week

Rdiger-36, 2025-02-25:
Its ready for testing in the new dev build version 1.0.4

heathen711, 2025-02-26:
Works.

Rdiger-36, 2025-03-02:
Its finally released int he latest build version 1.0.4.
I will close this Issue

closed 2025-03-02 by Rdiger-36