Rdiger-36/HaspelSync #29: Locations

heathen711, 2025-02-20:
In my other issue I requested for location of the spool to be added when it's in the AMS.

I tired to do it manually and the locations view is sub-optimal as they are not sorted and do not support some sense of grouping.

Maybe sorting would be enough to make it work, maybe need the ability to group locations or have sub-locations?

[image]

Rdiger-36, 2025-02-20:
Mh I don't really know what you mean with sub-locations.
What do you mean with sorting?

My first thoughts to this option is that the Printer Name, AMS and Slot will be filled in the "Location" filed.
For example like this "BambuLab P1S - AMS A - Slot 1" or "BambuLab P1S - AMS C- Slot 4".

ados8, 2025-02-21:
Is there plans for an AMS location and spools in the AMS are automatically assigned this location.
I don't know how you would handle the removal though without a database.
Example, PLA black with SN 1234 is put in the AMS.
Location changes from storage to AMS.
When roll is removed it would be good to have it revert back to its last location e.g. storage.

heathen711, 2025-02-21:
> What do you mean with sorting?

The location names on the Locations page are not sorted. Arrange them alphabetically? 

I noticed that I can drag them around to sort them... So maybe this isn't needed? but it could be nice to have the system sort automatically.

heathen711, 2025-02-21:
> My first thoughts to this option is that the Printer Name, AMS and Slot will be filled in the "Location" filed.
For example like this "BambuLab P1S - AMS A - Slot 1" or "BambuLab P1S - AMS C- Slot 4".

This will work perfectly. 

You said you were busy so I'll see if I can make you a patch to help.

heathen711, 2025-02-21:
@ados8 I need a third opinion:

Location per slot:

[image]

Or location per AMS: (I am thinking about adding the slot as a tag, so it's there in Spoolman if needed but when I'm looking in spoolman I tend to just want to know what is where so I know what I need to buy more of)

[image]

heathen711, 2025-02-21:
[image]

man adding an extra field was a PITA...

@Rdiger-36 you should know that your code currently overwrites all extra fields when you do a patch request. I've adjusted it in this patch.

> Note: This includes all the changes from my other issues, so it might need some adjusting, but it can be a starting point for you.

[heathen711-backend-with-location-and-slot-tagging.patch](https://github.com/user-attachments/files/18915476/heathen711-backend-with-location-and-slot-tagging.patch)

Rdiger-36, 2025-02-22:
Thank you very very much for helping me. I've read your patch and there are some good ideas and patches. I didnt even know that my version overwrite other tags, thanks for fixing this.
So your intention is not to put it in the "Locations" field?
Or what's your thoughts of your idea?

heathen711, 2025-02-22:
Yeah, I think grouping by AMS is more readable and using an extra field to indicate location keeps it clean but still usable.

Rdiger-36, 2025-02-22:
Yes, this sounds good.
The only question left is how to best manage it without a database, that the field is emptied again when the spool is unloaded.
I will think about it, but if you good an idea I'm open for it

heathen711, 2025-02-22:
Spoolman has a database, so while it would be resource intensive you could just clear all the location and slot info when you get an update, then reset it.

E.g. 

1. I have 3 things in AMS A
2. you get an update
3. clear everything from the AMS A and slot info
4. set the location and slot again.

Not ideal but would be good because it will ensure that things in the AMS are only the things in the AMS.

Because with the code above I still ran into having 7 things in AMS A when I changed out spools this morning.

We could also add a post update job to search for any spool with ams_slot that is not in an AMS location and clear it (for offline changes that the software missed)

Rdiger-36, 2025-02-22:
> We could also add a post update job to search for any spool with ams_slot that is not in an AMS location and clear it (for offline changes that the software missed)

I also thought about something like that, that would be the cleanest solution. The first idea would undesirably delete the location if someone have its own system

Rdiger-36, 2025-03-02:
I have been working on an optimal implementation and seamless post-processing for a week now. However, every approach I have pursued has led to further issues, such as handling offline printers that are then unloaded or problems with setting the tag, which would cause multiple cascading errors, especially in instances with multiple printers and AMS. Since I am not fully convinced by the current solution, I would prefer not to implement it.

The cleanest solution would be to implement a database, but I have not planned this for the app. It should merely provide a basic interface to Spoolman

heathen711, 2025-03-02:
Fair, but could you share some of the issues so we could talk about ways to work around them?

It may be that you’re missing something that a external person would see.

Rdiger-36, 2025-03-02:
Yes i can tell you about my problems but I didn't write them down and it was more a trial and error and there occur more and more problems of these. 
I will look through my history and will tell you about them.
It would be great if we could work together on this. 
If you’d like, you can code it yourself, and we can then see how it could be implemented and work perfectly.

heathen711, 2025-03-11:
So I took a stab at this, and make one with just location, ignoring the ams_slot.

Rdiger-36, 2025-03-11:
Yeah, that's exactly my point. I didn't have any time in the last few days, so I couldn't send you my results yet. But it seems like you totally get how tricky this is to handle.
Sometimes it feels like a good idea, but then you end up spending way too much time on it — and in the end, it's just not worth it.
Sometimes it's way easier to just do it the old-fashioned way, like manually putting the info about where the spool is located into the tag or the locations field.

Still, I want to sincerely thank you for your support and all the valuable input you've given to help move the project forward ❤️

closed 2025-03-11 by Rdiger-36