Rdiger-36/HaspelSync #30: Problem with sparkle Filament

karaktaka, 2025-02-20:
Hey, I have some Sparkle Filament in my AMS and it seems like there is an issue with creating/updating the spool in spoolman.

```
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:24:36 - P1S - Waiting for MQTT messages for Printer: 01P09C4A1400886...
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:25 - P1S - AMS [A] (hum: 5, temp: 0.0ºC)
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:25 - P1S -     - [A0] PLA Basic 000000FF (3%) [[ F9D10AE9AC20463AA2086BF8515A3FFF ]]
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:25 - P1S -         - Updated Spool-ID 39 => Black
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -     - [A1] PLA Matte FFFFFFFF (89%) [[ 48EF649D42C24312B0EAF10092C58F0D ]]
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -         - Updated Spool-ID 38 => Matte Ivory White
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -     - [A2] PLA Sparkle 483D8BFF (99%) [[ D610116DFAD64CDCAC1D3322A3368429 ]]
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -         - A new Filament and Spool can be created:
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -           Material: PLA, Color: Royal Purple Sparkle
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -           creating Filament and Spool...
bambu-bambulab-ams-spoolman-filamentstatus-1  | [ERROR] 2025-02-20_20:25:26 - P1S -           #####
bambu-bambulab-ams-spoolman-filamentstatus-1  | [ERROR] 2025-02-20_20:25:26 - P1S -           Filament creation failed:  Response code 500 (Internal Server Error)
bambu-bambulab-ams-spoolman-filamentstatus-1  | [ERROR] 2025-02-20_20:25:26 - P1S -           Error details:  HTTPError: Response code 500 (Internal Server Error)
bambu-bambulab-ams-spoolman-filamentstatus-1  | [ERROR] 2025-02-20_20:25:26 - P1S -           #####
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -     - [A3] Support for PLA 00000000 (20%) [[ 8BA24B5A578842D589C6E5B06F11BEDA ]]
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -         - Updated Spool-ID 35 => Support for PLA Black
bambu-bambulab-ams-spoolman-filamentstatus-1  | [LOG] 2025-02-20_20:25:26 - P1S -
```

The error in spoolman side is this one.

```
spoolman-spoolman-1  | uvicorn.error              ERROR    Exception in ASGI application
spoolman-spoolman-1  | Traceback (most recent call last):
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1967, in _exec_single_context
spoolman-spoolman-1  |     self.dialect.do_execute(
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 941, in do_execute
spoolman-spoolman-1  |     cursor.execute(statement, parameters)
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/sqlalchemy/dialects/mysql/aiomysql.py", line 95, in execute
spoolman-spoolman-1  |     return self.await_(self._execute_async(operation, parameters))
spoolman-spoolman-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/sqlalchemy/util/_concurrency_py3k.py", line 132, in await_only
spoolman-spoolman-1  |     return current.parent.switch(awaitable)  # type: ignore[no-any-return,attr-defined] # noqa: E501
spoolman-spoolman-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/sqlalchemy/util/_concurrency_py3k.py", line 196, in greenlet_spawn
spoolman-spoolman-1  |     value = await result
spoolman-spoolman-1  |             ^^^^^^^^^^^^
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/sqlalchemy/dialects/mysql/aiomysql.py", line 104, in _execute_async
spoolman-spoolman-1  |     result = await self._cursor.execute(operation, parameters)
spoolman-spoolman-1  |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/cursors.py", line 239, in execute
spoolman-spoolman-1  |     await self._query(query)
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/cursors.py", line 457, in _query
spoolman-spoolman-1  |     await conn.query(q)
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/connection.py", line 469, in query
spoolman-spoolman-1  |     await self._read_query_result(unbuffered=unbuffered)
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/connection.py", line 683, in _read_query_result
spoolman-spoolman-1  |     await result.read()
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/connection.py", line 1164, in read
spoolman-spoolman-1  |     first_packet = await self.connection._read_packet()
spoolman-spoolman-1  |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/connection.py", line 652, in _read_packet
spoolman-spoolman-1  |     packet.raise_for_error()
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/pymysql/protocol.py", line 219, in raise_for_error
spoolman-spoolman-1  |     err.raise_mysql_exception(self._data)
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/pymysql/err.py", line 150, in raise_mysql_exception
spoolman-spoolman-1  |     raise errorclass(errno, errval)
spoolman-spoolman-1  | pymysql.err.OperationalError: (1364, "Field 'id' doesn't have a default value")
spoolman-spoolman-1  |
spoolman-spoolman-1  | The above exception was the direct cause of the following exception:
spoolman-spoolman-1  |
spoolman-spoolman-1  | Traceback (most recent call last):
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/uvicorn/protocols/http/httptools_impl.py", line 409, in run_asgi
spoolman-spoolman-1  |     result = await app(  # type: ignore[func-returns-value]
spoolman-spoolman-1  |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/uvicorn/middleware/proxy_headers.py", line 60, in __call__
spoolman-spoolman-1  |     return await self.app(scope, receive, send)
[... 329 lines cut ...]
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/sqlalchemy/dialects/mysql/aiomysql.py", line 104, in _execute_async
spoolman-spoolman-1  |     result = await self._cursor.execute(operation, parameters)
spoolman-spoolman-1  |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/cursors.py", line 239, in execute
spoolman-spoolman-1  |     await self._query(query)
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/cursors.py", line 457, in _query
spoolman-spoolman-1  |     await conn.query(q)
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/connection.py", line 469, in query
spoolman-spoolman-1  |     await self._read_query_result(unbuffered=unbuffered)
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/connection.py", line 683, in _read_query_result
spoolman-spoolman-1  |     await result.read()
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/connection.py", line 1164, in read
spoolman-spoolman-1  |     first_packet = await self.connection._read_packet()
spoolman-spoolman-1  |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/aiomysql/connection.py", line 652, in _read_packet
spoolman-spoolman-1  |     packet.raise_for_error()
spoolman-spoolman-1  |   File "/home/app/spoolman/.venv/lib/python3.11/site-packages/pymysql/protocol.py", line 219, in raise_for_error
spoolman-spoolman-1  |     err.raise_mysql_exception(self._data)
[... cut ...]
spoolman-spoolman-1  | [SQL: INSERT INTO filament (registered, name, vendor_id, material, price, density, diameter, weight, spool_weight, article_number, comment, settings_extruder_temp, settings_bed_temp, color_hex, multi_color_hexes, multi_color_direction, external_id) VALUES (%s, %s, %s, %s, %s, %s, %s, %s, %s, %s, %s, %s, %s, %s, %s, %s, %s)]
spoolman-spoolman-1  | [parameters: (datetime.datetime(2025, 2, 20, 20, 25, 26), 'Royal Purple Sparkle', 5, 'PLA Sparkle', None, 1.26, 1.75, 1000.0, 250.0, None, None, 220, 60, '483D8B', None, None, 'bambulab_pla_royalpurplesparkle_1000_175_n')]
spoolman-spoolman-1  | (Background on this error at: https://sqlalche.me/e/20/e3q8)
```

Rdiger-36, 2025-02-22:
Mh it looks like there's in error on the spoolman database. 
Let me explain how this problem occur:
The service collects all data and send via API the request to create a new Filament with following stats:
```bash
[parameters: (datetime.datetime(2025, 2, 20, 20, 25, 26), 'Royal Purple Sparkle', 5, 'PLA Sparkle', None, 1.26, 1.75, 1000.0, 250.0, None, None, 220, 60, '483D8B', None, None, 'bambulab_pla_royalpurplesparkle_1000_175_n')]
```
In this step, spoolman will automatically register this filament and will automatically assign a id for it (normally it counts from 0 upwards).
But somehow spoolman couldn't create a new id for the new filament because of some inconsistent data in the database.

Unfortunately, I can't help you because it's an internal problem of spoolman.
As a workaround, you could try to create the filament manually in spoolman. The service should then recognize the filament, create a spool from it and connect it to the loaded spool from the AMS.
If this problem occur in the future with other new spools, then please try to contact the crew from spoolman

Rdiger-36, 2025-03-02:
Hi, do you got any updates for me?
If you won't answer I will close this issue in 3 days

closed 2025-03-11 by Rdiger-36