Rdiger-36/HaspelSync #33: Spoolman and MQTT Connection status: Disconnected

Rallyman03, 2025-02-21:
Service doesn't actually start up.

Server Log:
Log started at: 2025-02-21_17:32:59

[LOG] 2025-02-21_17:32:59 - Server - Version: 1.0.3

[LOG] 2025-02-21_17:32:59 - Server - Setting up configuration...

[ERROR] 2025-02-21_17:34:03 - Server - Failed to read log file for printerId "01P09A3B2400742": ENOENT: no such file or directory, open '/app/logs/01P09A3B2400742.log'

[ERROR] 2025-02-21_17:34:08 - Server - Failed to read log file for printerId "01P09A3B2400742": ENOENT: no such file or directory, open '/app/logs/01P09A3B2400742.log'

[ERROR] 2025-02-21_17:39:43 - Server - Error starting the service: RequestError: connect ETIMEDOUT 192.168.100.51:7912

[ERROR] 2025-02-21_17:40:37 - Server - Failed to read log file for printerId "01P09A3B2400742": ENOENT: no such file or directory, open '/app/logs/01P09A3B2400742.log'

--
Seems it's getting stuck there and not progressing.

heathen711, 2025-02-21:
Similar to: https://github.com/Rdiger-36/bambulab-ams-spoolman-filamentstatus/issues/24

I added a volume for the logs and manually created the file to see it.
`- ./config/printers/logs:/app/logs`
Not a solution just a workaround.

Rallyman03, 2025-02-21:
Thank you. 

I've added this to my docker compose:
    volumes:
      - /home/spirit/spoolman/filamentstatus/config/printers:/app/printers
      - /home/spirit/spoolman/filamentstatus/config/printers/logs:/app/logs

And manually created 01P09A3B2400742.log in the config/printers/logs folder but I'm still getting this error:

[ERROR] 2025-02-21_22:02:47 - Server - Failed to read log file for printerId "01P09A3B2400742": ENOENT: no such file or directory, open '/app/logs/01P09A3B2400742.log'

What am I mising?

Rdiger-36, 2025-02-22:
@heathen711 thanks for your help with the workaround. I will take a look why the logging fails. 

@Rallyman03 
The timeout could occur because spoolman is not reachable from inside the docker container. 
Are the spoolman and bambulab-ams-spoolman-filamentstatus containers on the same network and host?
Then please try this:
- SPOOLMAN_IP=spoolman
- SPOOLMAN_PORT=8000

Rallyman03, 2025-02-22:
> [@heathen711](https://github.com/heathen711) thanks for your help with the workaround. I will take a look why the logging fails.
>
> [@Rallyman03](https://github.com/Rallyman03) The timeout could occur [...]

Rdiger-36, 2025-02-22:
That means it works now?

Rallyman03, 2025-02-22:
No. I was responding to you saying there weren't on the same network. They are on the same network and I can ping the spoolman container from the filamentstatus container. So that is not the issue.

Rdiger-36, 2025-02-22:
Ok, I will do what I can to make in run successfully. I will reopen the issue and add some things to debug the problem in a new dev build if you want

Rallyman03, 2025-02-22:
That would be great. Please let me know if you need anything from me.

Rdiger-36, 2025-02-23:
I have done some research. Can you please connect to your bambulab-ams-spoolman-filamentstatus container with this command:
```bash
docker exec -it bambulab-ams-spoolman-filamentstatus /bin/sh
```
Then please test if you can reach your spoolman container port:
```bash
nc -zv $SPOOLMAN_IP $SPOOLMAN_PORT
```

it should look like this (test on my machine):

```bash
/app # nc -zv $SPOOLMAN_IP $SPOOLMAN_PORT
192.168.1.9 (192.168.1.9:30010) open
```

If that's all true, make sure that the spoolman container is up and healthy and then start the bambulab-ams-spoolman-filamentstatus container

Rallyman03, 2025-02-23:
I tried the command and was getting no return. So I double checked to see if the linux firewall was blocking that port. Turns out it was. As soon as I opened it and restarted the container.  Thanks for your help, you can close the issue.

Rdiger-36, 2025-02-23:
That sounds good. No problem, we all want that it will work properly.

closed 2025-02-23 by Rdiger-36