Rdiger-36/HaspelSync #35: No spools are created in spoolman

marcokreeft87, 2025-03-12:
I started the container, everything seems to be fine. But only the field is created in spoolman, no spools 

`[LOG] 2025-03-12_15:54:36 - Server - Version: 1.0.4
[LOG] 2025-03-12_15:54:36 - Server - Setting up configuration...
[LOG] 2025-03-12_15:54:36 - Server - Starting service...
[LOG] 2025-03-12_15:54:36 - Server - Backend running on http://localhost:4000
Server Server Log file created
[LOG] 2025-03-12_15:54:36 - Server - Spoolman connected successfully!
[LOG] 2025-03-12_15:54:36 - Server - Checking Vendors...
[LOG] 2025-03-12_15:54:36 - Server - Vendor "Bambu Lab" exists: true
[LOG] 2025-03-12_15:54:36 - Server - Checking Extra Field "tag"...
[LOG] 2025-03-12_15:54:36 - Server - Spoolman Extra Field "tag" for Spool is set: true
[LOG] 2025-03-12_15:54:36 - Server - Backend running on http://localhost:4000
[LOG] 2025-03-12_15:54:36 - X1C - MQTT not running for Printer: xxxx, attempting to reconnect...
[LOG] 2025-03-12_15:54:36 - X1C - Setting up MQTT connection for Printer: xxxx...
[LOG] 2025-03-12_15:54:36 - X1C - MQTT client connected for Printer: xxxx
[LOG] 2025-03-12_15:54:36 - X1C - Waiting for MQTT messages for Printer: xxxx...
[LOG] 2025-03-12_15:54:36 - X1C - AMS [A] (hum: 5, temp: 24.2ºC)
[LOG] 2025-03-12_15:54:36 - X1C -     - [A0] PLA 898989FF [[ N/A ]]
[LOG] 2025-03-12_15:54:36 - X1C -     - [A1] PLA AF7933FF [[ N/A ]]
[LOG] 2025-03-12_15:54:36 - X1C -     - [A2] PLA FFFFFFFF [[ N/A ]]
[LOG] 2025-03-12_15:54:36 - X1C -     - [A3] PLA 161616FF [[ N/A ]]
[LOG] 2025-03-12_15:54:36 - X1C - 
`
It just keeps repeating te last 6 lines every sec

Rdiger-36, 2025-03-12:
> They are Sunlu spools, does that matter?

Yes, only BambuLab Spools are supported because of their RFID chips. I don't know why, but I thought I had written it in the README, but it doesn't seem so. I will add it.
Unfortunately, there is no possibility to use 3rd party spools because they cannot communicate with the AMS like you see in your logs.

weakfl, 2025-03-13:
@Rdiger-36 hope you don't mind me mentioning a complementary solution from another developer...

@marcokreeft87 you could take a look at https://github.com/drndos/openspoolman. But note that there is an open PR with some vital bugfixes.

Personally I'm using both, and the solutions generally don't interfere.

Rdiger-36, 2025-03-13:
Hi @weakfl that’s absolutely no problem at all, and I really appreciate supporting other projects and creators as well. I’ve even referenced this project in my own issues before — I simply forgot to mention it in my comment.
After all, we’re all part of the same community, trying to find the system that works best for each of us.

marcokreeft87, 2025-03-13:
I started my own solution with Home Assistant and a self made API that translates and updates spoolman

marcokreeft87, 2025-03-14:
So for both bambu and non bambu spools I've created a api https://github.com/marcokreeft87/spoolman-updater

closed 2025-03-14 by marcokreeft87