Rdiger-36/HaspelSync #36: PLA Basic (Several Colours) Breaks Identification

Xirii, 2025-03-15:
Managed to get everything talking and working great, however as I started inventory I found that after loading into the AMS, some spools break processing - manually creating them by taking the serial number from BambuStudio and inputting into Spoolman gets everything going again, so this may be related to error detection. 

So far the following filaments break the script:
https://uk.store.bambulab.com/products/pla-basic-filament?id=41465098960956
https://uk.store.bambulab.com/products/pla-basic-filament?id=41465096077372
https://uk.store.bambulab.com/products/pla-basic-filament?id=40206189264956
https://uk.store.bambulab.com/products/pla-silk?id=40612838637628
https://uk.store.bambulab.com/products/pla-basic-filament?id=41465093718076

It doesn't appear to be mismatched colours (at a surface level), as the PLA Basic Magenta colour code matched Bambu's records.

[image]

```plaintext
[LOG] 2025-03-15_17:47:25 - BambuLab X1 Carbon -

[LOG] 2025-03-15_17:47:54 - BambuLab X1 Carbon - AMS [A] (hum: 5, temp: 27.7ºC)

[LOG] 2025-03-15_17:47:55 - BambuLab X1 Carbon - - [A0] PLA Silk 0047BBFF (83%) [[ BE2378CA58C948BFA01F4E200EC0FA41 ]]

[LOG] 2025-03-15_17:47:55 - BambuLab X1 Carbon - - Updated Spool-ID 14 => Neon City (Blue-Magenta)

[LOG] 2025-03-15_17:47:55 - BambuLab X1 Carbon - - [A1] PLA Basic 5E43B7FF (81%) [[ 7233F31C60BF4689843EA850C0594E23 ]]

[ERROR] 2025-03-15_17:47:55 - BambuLab X1 Carbon - Error processing message for Printer: 00M09C441101981 - Cannot read properties of undefined (reading 'toLowerCase')

[LOG] 2025-03-15_17:48:25 - BambuLab X1 Carbon - AMS [A] (hum: 5, temp: 27.7ºC)

[LOG] 2025-03-15_17:48:25 - BambuLab X1 Carbon - - [A0] PLA Silk 0047BBFF (83%) [[ BE2378CA58C948BFA01F4E200EC0FA41 ]]

[LOG] 2025-03-15_17:48:25 - BambuLab X1 Carbon - - Updated Spool-ID 14 => Neon City (Blue-Magenta)

[LOG] 2025-03-15_17:48:25 - BambuLab X1 Carbon - - [A1] PLA Basic 5E43B7FF (81%) [[ 7233F31C60BF4689843EA850C0594E23 ]]

[ERROR] 2025-03-15_17:48:25 - BambuLab X1 Carbon - Error processing message for Printer: 00M09C441101981 - Cannot read properties of undefined (reading 'toLowerCase')
```

After removing the spool or manually configuring in Spoolman, everything works as expected:

[image]

```plaintext
[LOG] 2025-03-15_17:48:55 - BambuLab X1 Carbon - AMS [A] (hum: 5, temp: 27.7ºC)

[LOG] 2025-03-15_17:48:55 - BambuLab X1 Carbon - - [A0] PLA Silk 0047BBFF (83%) [[ BE2378CA58C948BFA01F4E200EC0FA41 ]]

[LOG] 2025-03-15_17:48:55 - BambuLab X1 Carbon - - Updated Spool-ID 14 => Neon City (Blue-Magenta)

[LOG] 2025-03-15_17:48:56 - BambuLab X1 Carbon - - [A1] N/A N/A (0%) [[ N/A ]]

[LOG] 2025-03-15_17:48:56 - BambuLab X1 Carbon - - [A2] PLA Basic FF6A13FF (61%) [[ 24D90B43B96B4C78B3DD655D0DD8B5D9 ]]

[LOG] 2025-03-15_17:48:56 - BambuLab X1 Carbon - - Updated Spool-ID 3 => Orange

[LOG] 2025-03-15_17:48:56 - BambuLab X1 Carbon - - [A3] PLA Basic 000000FF (95%) [[ D05C9281C8CA4B51B5F602172C056FA2 ]]

[LOG] 2025-03-15_17:48:56 - BambuLab X1 Carbon - - Updated Spool-ID 4 => Black
```

Rdiger-36, 2025-03-16:
Hi, can you please send me your collected data from your printer?
With that I can debug the problems.

You can use the build in debug-printers command.
Please let me know if you need help.

Rdiger-36, 2025-03-17:
I found the bug. I will release the new Version 1.0.5 soon

Xirii, 2025-03-19:
Apologies it looks like I've not been getting email notifications from GitHub! I shall delete a known "bad" spool and get back to you.

Xirii, 2025-03-19:
Tested all the spools and those which weren't tested originally and all is working as expected. Thanks for your hard work!