Rdiger-36/HaspelSync #41: Wrong remaining weight for 0.5kg filaments (like supports).

tomeko12, 2025-06-09:
The remaining filament weight on 0.5 kg spools (like support filaments) is reported incorrectly. 
The AMS shows the remaining filament percentage relative to a full 1 kg spool. My new **Support for PLA** filament (sold on a 0.5 kg spool) is reported as 205g/500g (41%), but in reality, it's around 500g (I believe the 10% discrepancy is a measurement error).
```json
{
              "bed_temp": "0",
              "bed_temp_type": "0",
              "cali_idx": -1,
              "cols": [
                "FFFFFFFF"
              ],
              "ctype": 0,
              "drying_temp": "55",
              "drying_time": "8",
              "id": "3",
              "nozzle_temp_max": "240",
              "nozzle_temp_min": "190",
              "remain": 43,
              "state": 11,
              "tag_uid": "33..............................00",
              "total_len": 330000,
              "tray_color": "FFFFFFFF",
              "tray_diameter": "1.75",
              "tray_id_name": "S02-W1",
              "tray_info_idx": "GFS02",
              "tray_sub_brands": "Support for PLA",
              "tray_type": "PLA-S",
              "tray_uuid": "EE1........................................B",
              "tray_weight": "500",
              "xcam_info": "000000000000000000000000"
            }
```

[image]

tomeko12, 2025-06-10:
I've checked some others non 1kg Bambu spools:
- PLA-CF (250g spool added to printer): 437g gross / ca. 194g net
- Support for PLA/PETG (250g spool added to priner): 481g gross / ca.  238g net
- Support for ABS (strandard 0.5kg spool): 755g gross / ca. 502g net
- Support for PA/PET (standard 0.5kg spool): 763g gross / ca. 510g net
[image]
```json
          "tray": [
            {
              "bed_temp": "0",
              "bed_temp_type": "0",
              "cali_idx": -1,
              "cols": [
                "000000FF"
              ],
              "ctype": 0,
              "drying_temp": "55",
              "drying_time": "8",
              "id": "0",
              "nozzle_temp_max": "240",
              "nozzle_temp_min": "210",
              "remain": 12,
              "state": 11,
              "tag_uid": "63...................00",
              "total_len": 330000,
              "tray_color": "000000FF",
              "tray_diameter": "1.75",
              "tray_id_name": "A50-K0",
              "tray_info_idx": "GFA50",
              "tray_sub_brands": "PLA-CF",
              "tray_type": "PLA-CF",
              "tray_uuid": "AB............................5D7",
              "tray_weight": "250",
              "xcam_info": "8..................3F"
            },
            {
              "bed_temp": "0",
              "bed_temp_type": "0",
              "cali_idx": -1,
              "cols": [
                "00000000"
              ],
              "ctype": 0,
              "drying_temp": "70",
              "drying_time": "8",
              "id": "1",
              "nozzle_temp_max": "220",
              "nozzle_temp_min": "190",
              "remain": 20,
              "state": 11,
              "tag_uid": "5...........................0",
              "total_len": 330000,
              "tray_color": "00000000",
              "tray_diameter": "1.75",
              "tray_id_name": "S05-C0",
              "tray_info_idx": "GFS05",
              "tray_sub_brands": "Support for PLA",
              "tray_type": "PLA-S",
              "tray_uuid": "6D280E..................345",
              "tray_weight": "250",
              "xcam_info": "A4...................................3F"
            },
            {
              "bed_temp": "0",
              "bed_temp_type": "0",
              "cali_idx": -1,
              "cols": [
                "FFFFFFFF"
              ],
              "ctype": 0,
              "drying_temp": "55",
              "drying_time": "8",
              "id": "2",
              "nozzle_temp_max": "270",
              "nozzle_temp_min": "240",
              "remain": 52,
              "state": 11,
              "tag_uid": "A...........................0",
              "total_len": 330000,
              "tray_color": "FFFFFFFF",
              "tray_diameter": "1.75",
              "tray_id_name": "S06-W0",
              "tray_info_idx": "GFS06",
              "tray_sub_brands": "Support for ABS",
              "tray_type": "ABS-S",
              "tray_uuid": "47..............................94",
              "tray_weight": "500",
              "xcam_info": "A..............................3F"
            },
            {
              "bed_temp": "0",
              "bed_temp_type": "0",
              "cali_idx": -1,
              "cols": [
                "C0DF16FF"
              ],
              "ctype": 0,
              "drying_temp": "80",
              "drying_time": "12",
              "id": "3",
              "nozzle_temp_max": "300",
              "nozzle_temp_min": "280",
              "remain": 53,
              "state": 11,
              "tag_uid": "9..........................................0",
              "total_len": 330000,
              "tray_color": "C0DF16FF",
              "tray_diameter": "1.75",
              "tray_id_name": "S03-G1",
              "tray_info_idx": "GFS03",
              "tray_sub_brands": "Support for PA",
              "tray_type": "PA-S",
              "tray_uuid": "B............................................C",
              "tray_weight": "500",
              "xcam_info": "10.........................................3F"
            }
          ]

```

Teh-Hamstah, 2025-06-13:
I've also just noticed that the numbers set on Spoolman seem a bit odd. The spool displays correctly as 250g in the list view (at least as far as the 100% remaining is being incorrectly reported):

[image]

However, the initial weight and the filament record both show the weight as 1000g:

[image]

[image]

This could be related to the fact that this is a very early spool from Bambu (I got this with my X1C in January 2023) or that it was a sample spool, but I'm not sure. The MQTT tray weight correctly reports it as 250g.

Rdiger-36, 2025-06-13:
@Teh-Hamstah 
The filament and the spool are treated as separate entries.
Each filament type always has an initial weight of 1000g.
However, when a new spool is created, its initial weight is set based on the tray_weight value sent by the spool’s MQTT data.
This way, you can have any filament type combined with a spool of any individual weight.
For example, Bambu Lab gives you a sample spool of filament for free when you buy a printer. So there can be different spool weights, but the filament itself is always theoretically the same.

> Scratch the above - I was reading the 34 as a percentage, but I just read the above comments, went and weighed the spool manually, and found that it was reporting 340g on a 250g spool because the spool was slipping when it was trying to calculate the remaining weight.
>
>I just put the spool into a different AMS and it measured it as 210g, which the AMS dashboard correctly reports.

This means that this is "solved"?

Rdiger-36, 2025-06-13:
Bambu measures the remaining filament by counting how often the RFID chip passes the sensor.
A full 1000g spool will result in fewer passes, while a nearly empty spool (e.g. 200g left) will trigger more frequent passes.

This method is quite inaccurate, but we don’t have any better way to report the remaining filament.

It would be possible to read the print job data and build some logic to calculate the required filament for each print. However, this gets complicated if you have to account for failed or aborted prints, as well as multi-color prints.
That would require a lot of additional effort, and unfortunately, I don’t currently have the time or resources to implement such a solution.

Teh-Hamstah, 2025-06-13:
> This means that this is "solved"?

The interface is correctly reporting the remaining weight on the spool (minus inaccuracies due to the way they measure the remaining amount, which as you say, can't be helped, but is close enough). What I was originally stating about it reporting the wrong value was my error.

The only issue I see is that the "initial weight" value that was sent to Spoolman was 1000g, which for this spool isn't correct. The initial weight on this spool was 250g, as reported by the tray weight. That's why I mentioned the filament record, as that's why the filament is named "Bambu Lab - Support for PLA White (Support W, 1.75 mm, 1 kg)" when it should really be "Bambu Lab - Support for PLA White (Support W, 1.75 mm, 250 g)"

Rdiger-36, 2025-06-13:
I understand what you mean, this relates to Spoolman. They have many filament profiles that you can use to create a new filament entry.

I use every bit of information available from a loaded spool and try to find a matching filament profile to create a new one in Spoolman.

As far as I can see, there is no profile for this filament with only 250g.

Teh-Hamstah, 2025-06-13:
Fair enough - honestly it was a very minor point since your code does get Spoolman to correctly report the actual amount used and remaining even though the initial amount is different. Happy to call that resolved 😄

Rdiger-36, 2025-06-13:
Yeah, there are some minor things that make it a bit tricky, but I do what I can and I think it does what it should now.
Thank you for helping me to make it better!

Rdiger-36, 2025-06-13:
The latest version 1.0.7 is now released and implemented this feature