Rdiger-36/HaspelSync #43: Utilization not updating

Peabuttt, 2025-07-25:
Hey! i just finished setting up everything. The spools are identified, created and added to spoolman. But, they are not updated during or after the print. The timestamp is updated, but not utilization.

Im using AMS 2 pro. Could that be the issue?

Rdiger-36, 2025-07-30:
Hi, I’ve been very busy the last few weeks and days.

Thank you for your research and letting me know about your results. The main problem with Bambu is that the estimated filament depends on how often the RFID chip passes the sensor. Sometimes it doesn’t function correctly, causing issues.

You’re right. If you haven’t printed much, it could be that the filament won’t update and remain at its current percentage.

If you want to increase output, you can use the built-in debug-printers command via the command line. This allows you to view everything the printer sends. Additionally, you can run the container with the debug environment variable to gain more insights into the process behind updating the spools. Furthermore, if you enter developer mode in your browser, you can check all API calls that are working with both the back-end and front-end. If you have any further questions, please let me know.

Rdiger-36, 2025-08-24:
Hey @Peabuttt do you tested it for a while?
Does it work properly?

If you don't reply in the next 3 days, I will close this issue.