Rdiger-36/HaspelSync #44: Different spools merged

junkman690, 2025-07-31:
2 spools of the same type/colour were merged despite having the tag already set.

> [LOG] 2025-07-27_04:15:09 - P1S - - [A2] PLA Basic 000000FF (2%) [[ D1BB3F5B0B804F3E938A07954007E894 ]]
> 
> [LOG] 2025-07-27_04:15:09 - P1S - - Updated Spool-ID 3 => Black
> 
> [LOG] 2025-07-30_13:04:00 - P1S - - [A2] PLA Basic 000000FF (100%) [[ C9EB3227173D49F69BAD0F1AD4BA8F3C ]]
> 
> [LOG] 2025-07-30_13:04:00 - P1S - - Found mergeable Spool => Spoolman Spool ID: 3, Material: PLA Basic, Color: Black
> 
> [LOG] 2025-07-30_13:04:00 - P1S - merging Spool...
> 
> [LOG] 2025-07-30_13:04:00 - P1S - Spool successfully merged with Spool-ID 3 => Black

#37 and #16 look like this has been identified and resolved however it is present in latest v.1.0.8

Rdiger-36, 2025-07-31:
Hi, did you use the environment variable  "NEVER_MERGE_IF_TAG" and set it to true?

junkman690, 2025-07-31:
Missed that one despite reading through the docs a few times. I've set it now and imagine all will be now well for the next spool. Thanks for the project and quick reply!

closed 2025-07-31 by junkman690